raspberrypi/rpi-imager #10: Windows SmartScreen blocking installation

shatteredsword, 2020-03-05:
Windows 10.0.18363 Build 18363

[image]

ghollingworth, 2020-03-05:
Click on More Info and then click Run...  You can check that it has been signed by us using our registered certificate.  Microsoft want you to pay them lots of extra money to get rid of this screen, although it should go away after a few people install it.

closed 2020-03-05 by ghollingworth

shatteredsword, 2020-03-05:
I'm aware of how to get around this issue, but the issue still exists. By closing this you are just making it more difficult for less technical people to get help on this issue...

ghollingworth, 2020-03-05:
As I said, this is unavoidable, and a normal part of code signing with a standard certificate (which is what we have...), next time we'll have to buy an Extended Validation certificate to get rid of it.

From 'the web':

If you have a standard code signing certificate, some time will be needed for your application to build trust. Microsoft affirms that an Extended Validation (EV) Code Signing Certificate allows to skip this period of trust building. According to Microsoft, extended validation certificates allow the developer to immediately establish reputation with SmartScreen. Otherwise, the users will see a warning like "Windows Defender Smartscreen prevented an unrecognized app from starting. Running this app might put your PC at risk.", with the two buttons: "Run anyway" and "Don't run".

Another Microsoft resource states the following (quote): "Although not required, programs signed by an EV code signing certificate can immediately establish reputation with SmartScreen reputation services even if no prior reputation exists for that file or publisher. EV code signing certificates also have a unique identifier which makes it easier to maintain reputation across certificate renewals."