raspberrypi/rpi-imager #8: Error downloading OS list from Internet when running from rpi4

camccar, 2020-03-05:
Tried out the new imagewritter this morning on a raspberry pi 4 running raspbian and get the following error. 

[image]

maxnet, 2020-03-05:
If you need a temporary workaround (before this is addressed server side), adjust Raspbian's security policies: https://www.debian.org/releases/stable/i386/release-notes/ch-information.en.html#openssl-defaults

ghollingworth, 2020-03-05:
As maxnet notes, this is actually a server problem on the downloads.raspberrypi.org site, we're getting it fixed ASAP.  So the problem should go away (although it theoretically should work one time in 9 because one of the load balancers has actually been updated!)

camccar, 2020-03-06:
Tested this morning and it was working. Gonna go ahead and close this.

closed 2020-03-06 by camccar

ElApe, 2021-06-12:
June 12th 2021 and still seeing this issue, I've never actually seen it work. Could it possibly be to do with limited disk space on the host device? I currently only have 18Gb free of 180Gb on my SSD drive running Windows 10

maxnet, 2021-06-12:
> Could it possibly be to do with limited disk space on the host device?

No, does not consume more than a couple kilobytes at this stage.

Any kind of network connectivity or network configuration problem can also cause this message.
E.g. if you do have configured an IPv6 address, but IPv6 is not acutally working you can also get this message.
Or if there is a proxy server listed in your Windows internet settings, but that proxy server is not working without extra authentication.

ElApe, 2021-06-12:
No IPV6 on my network and no proxy settings either.

Everything else is able to access the internet fine, it's just the imager that gives me issues. Is there a log file anywhere?

⁣Regards

Nick Sharp​

maxnet, 2021-06-12:
>Is there a log file anywhere?

Nope
(--debug output does not cover the OS list download performed in qml)

maxnet, 2021-06-12:
Hmm, thinking about it. 
There do is one way you may be able to get a little bit more log information about the OS list download.

Open an administrative command prompt (otherwise you cannot set environment variables for privileged applications), by going to the Windows start menu, find "Command Prompt", right-click it, select "run as administrator".

type:

```
cd \program files (x86)\raspberry pi imager
set QML_XHR_DUMP=1
rpi-imager --debug
```

Lines that start with XMLHttpRequest may be of interest.

ElApe, 2021-06-13:
Hi Maxnet,
thanks for replying, this is the output I see...
C:Program Files (x86)Raspberry Pi Imager>set QML_XHR_DUMP=1

C:Program Files (x86)Raspberry Pi Imager>rpi-imager --debug

C:Program Files (x86)Raspberry Pi Imager>DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Roboto", lfWidth=0, lfHeight=-11) dpi=96
DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Noto", lfWidth=0, lfHeight=-11) dpi=96
XMLHttpRequest: GET https://downloads.raspberrypi.org/os_list_imagingutility_v2.json
XMLHttpRequest: ERROR https://downloads.raspberrypi.org/os_list_imagingutility_v2.json
 QNetworkReply::RemoteHostClosedError RemoteHostClosedError ""

If I paste the URL into a browser I can pull down the JSON file. 

Some more info on my config

Fails on...
Windows 10 20H2 release. 
FSecure Computer Protection 21.5, have tried with and with out it installed.
Windows Firewall has been disabled in case this was the issue.

Proxy Settings...
IPV6 Status.
Local DHCP RPI 3B+ ISC-DHCP-SERVER
Local DNS RPI 3B+ Bind9
Forwarders Quad 9 and 8.8.8.8

Network...
192.168.1.0/24
Cisco SG30028 Switch
Cisco C887VA-K9 Router
Zen Internet 60Mbs FTTC

On my Raspberry Pi 400 the json file is downloaded fine.

It's no great shakes as I can download the .img files I need manually and image them or use rpi-immager on one of my Pi's but I I think it may help others using Windows boxes who may have similar issues.

If you require any further information please let me know 

Regards

Nick Sharp

ElApe, 2021-06-13:
Sorry didn't take across the embedded images for the proxy settings and IPV6 which basically shows they are all disabled.

lurch, 2021-06-13:
> Sorry didn't take across the embedded images

You can attach images to a GitHub issue using the GitHub webinterface, you just can't do that when replying to an issue via email.

maxnet, 2021-06-13:
> Sorry didn't take across the embedded images for the proxy settings and IPV6 which basically shows they are all disabled.

Recall in the Internet options where you can click on the "proxy settings" button there is also a checkbox somewhere to auto-detect settings, make sure that is disabled as well.
As it is not obvious in the GUI whether it detected a proxy to use. (Do can see it commands like "netsh winhttp show proxy")

> QNetworkReply::RemoteHostClosedError RemoteHostClosedError ""

Most problems with Windows' own firewall and proxies that are not reachable at all should give UnknownNetworkError instead though.
Not sure what error third-party security software (like anti-virus, firewalls and parental control stuff) would give.

Don't know what is wrong.

ElApe, 2021-06-13:
Hi Maxnet,

C:UsersNSharp>netsh winhttp show proxy

Current WinHTTP proxy settings:

 Direct access (no proxy server).

No proxy settings as you can see. Also I have tried with/without firewall enabled and with/without F-Secure so I'm pretty sure its not related to that.

As I can download the JSON file from a browser it would suggest to me its not a setting issue on my PC though.

It's a weird one isn't it?

maxnet, 2021-06-13:
>As I can download the JSON file from a browser it would suggest to me its not a setting issue on my PC though.

Hard to tell.
You did mention it did work on a Pi 400, so that would rule out any network level problems at you, your provider, and our hosting provider, and point to your PC instead :-)

Some browsers may also be more forgiving to some errors.
downloads.raspberrypi.org resolves to several different servers at 12 IPv4 addresses and 12 IPv6 addresses.

Whether it will try reconnecting to a different address if first attempt failed is application/browser dependent.
So sometimes configuration issues (such as when the computer has an IPv6 address, while IPv6 is not functional) do not show in all applications.

That said, I do not have a clue what is exactly wrong in your case.
And it is kinda hard to give support for things that are not reproducible here.
(There may be other things that could be tried to debug this, like packet sniffing. But if you haven't done it before, and do not know what to look for, I wouldn't advice end-users to do that.)

ElApe, 2021-06-13:
I'll fire up wireshark tomorrow and see what's occuring. 

⁣Regards

Nick Sharp​