raspberrypi/rpi-imager #9: Allow installing Home Assistant

balloob, 2020-03-05:
Opening an issue as per https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/#comment-1523756

[Home Assistant](https://www.home-assistant.io/) is an open source home automation platform that focuses on local control and privacy. We provide our own operating system that turns your Raspberry Pi into a home automation hub, including easy updates via the UI. We recommend users that are starting out to get a Raspberry Pi 4. We support all variants of the Raspberry Pi. 

It would be great if Home Assistant would be supported by the Raspberry Pi Imager.

balloob, 2021-05-07:
@maxnet is there anything we can do to help move this forward?

ghollingworth, 2021-05-07:
You'll need to provide a .json snippet for your images...

As per the top level .json

http://downloads.raspberrypi.org/os_list_imagingutility_v2.json

You can test your own changes using "rpi-imager --repo filename.json" on the command line

agners, 2021-05-07:
@ghollingworth should we store that json snipet online somewhere? Are you going to update automatically if we maintain that using our release pipeline?

balloob, 2021-05-07:
@agners looking at the entry for OctoPi, we just generate a JSON and add a reference in the main listing:

[image]

https://unofficialpi.org/rpi-imager/rpi-imager-octopi.json

agners, 2021-05-07:
Oh I see, somehow did not notice that this is referencing jsons. I guess Home Assistant would go under "Other specific purpose OS":

```
               {
                    "name": "Home Assistant",
                    "description": "Open-source home automation platform with a focus on local control and privacy.",
                    "icon": "https://version.home-assistant.io/rpi-imager-haos.png",
                    "subitems_url": "https://version.home-assistant.io/rpi-imager-haos.json"
               },
```

The icon/json should be up soon.

agners, 2021-05-24:
It seems the image list made it in, Home Assistant is now available in the official repo! :tada: 

I think with that this can be closed.

closed 2021-05-24 by balloob

balloob, 2021-05-24:
🎉 

https://twitter.com/home_assistant/status/1396877204026363905

b-reich, 2021-12-23:
> @ghollingworth should we store that json snipet online somewhere? Are you going to update automatically if we maintain that using our release pipeline?

Something with your release pipeline wont work. It's still on 6.2.

agners, 2021-12-23:
It is a manual step currently. Updated to release 7.0.