raspberrypi/rpi-imager #14: Windows10 : Poor Scaling, Small font

bsimmo, 2020-03-06:
Under Windows 10, it doesn't respond to OS "Make Everything Bigger" scaling very well.
It ignores Font scaling "Make Text Bigger"

The default roboto, size 10/12 fonts are hard to read on 15.6" 1080 displays standard laptop displays. Many displays are smaller then this at 13"/14" with 1080p

Make them 16point minimum, I would guess.
Allow the program to respond to OS settings.

How, I don't know. though I can see the font sizes in the files.

Refrence https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=267080#p1622462

maxnet, 2020-03-06:
Strange.

It is not that bad on smaller displays.
This is how it looks like on the average 11" 1080p $ 300 Chinese Teclast laptop, without any special settings.

[image]

bsimmo, 2020-03-06:
what's you scaling and font size set to.  My Win10 new setup defaults to 125%?

maxnet, 2020-03-06:
Window's default, which is 150% scaling for my display. No accessibility settings (smallest font size).

It is quite possible that it does not know anything about Windows specific features like font sizing (it's written with a cross-platform toolkit, and such things are not universal), but it shouldn't be that small in the first place.
It do should adjust size a bit based on dpi reported by monitor, so perhaps something is strange there.

bsimmo, 2020-03-08:
Tried it on my PC and with different monitors and with them attached to the laptops.  At 100%/100% it is the size you would expect, It is still a small Font as it Roboto at 10/12 from what I can see and is a bit smaller than the default (Arial I think on Windows10, it's all hidden nowadays) and the harder to read part is in bold.
Bump it up to 16 (some default Web page guidelines iirc, especially for mobile devices.)
If you remove the hard set font size, what does that do, is there a scaling size option ?

Then start being OS compatible with font, scaling and readers.
This is RPF  accessibility under their statements.

If I find it hard to read, then it is going to be harder for older, partial sighted and the contrast is very poor on the lightweight test.  

SIDE not : _BUT the program works (mostly if you read the forum, some bugs noticed) and should be great to build on :-)_

maxnet, 2020-03-08:
>If you remove the hard set font size, what does that do, is there a scaling size option ?

There is no easy button for that.

But for educational purposes:
- go to the Windows start menu, find "Command Prompt", right-click it, select "run as administrator"
- type:

```
cd \program files (x86)\raspberry pi imager
set QT_SCALE_FACTOR=1.6
imagingutility
```

It will show you what it would like if we made the application 160% larger, which is essentially what you are proposing by saying fonts should be at least 16 points. (Just enlaring fonts without other parts of application is a bit problematic)
For some displays that will essentially mean it will be a full screen application.
Not sure if that is something the design guys will agree with...

(reason you need to run command prompt as administrator, is because Windows will refuse to pass environment variables to UAC application that need more privileges otherwise)

bsimmo, 2020-03-08:
No reason it cannot be a proper scrollable, resizable  full screen program.
Pointless using fixed window sizes like it is at the moment.

Use the flexibility of a modern system.

But that may just be me (you don't see chrome, office programs, steam etc
using small fixed windows.)

maxnet, 2020-03-08:
Text programs and browsers can do word wrapping if the text does not fit on a single line.
With the (listview style) components used in application like this, things are not that simple...

andrum99, 2020-03-11:
I'm seeing this as well. On my system it is not a problem, since my display is only 1080p which defaults to a 125% scale factor. Therefore there isn't much of a difference between the scaled and unscaled size.

This is going to cause issues for anyone running the tool on a very high resolution display, e.g. 4K.

maxnet, 2020-03-11:
>This is going to cause issues for anyone running the tool on a very high resolution display, e.g. 4K.

Qt also takes the DPI reported by your screen into consideration, not just what scale factor you set.
So hard to tell if you do not actually have a 4k display.

bsimmo, 2020-03-11:
It only started to scale (in reality) at 175% for me.

Wouterdek, 2020-04-21:
The issue here is fractional scaling. I have made screenshots at 100%, 150% and 200% scaling. Notice how 100% and 200% are fine, but 150% is too small.

[100%](https://user-images.githubusercontent.com/913101/79886936-36a3f980-83fa-11ea-8be6-41f290ca9d9d.png)
[150%](https://user-images.githubusercontent.com/913101/79886946-3a378080-83fa-11ea-91b2-38f9cb20f2bf.png)
[200%](https://user-images.githubusercontent.com/913101/79887677-8d5e0300-83fb-11ea-9a70-57501efee476.png)

The source of this bug is scaling in Qt. To fix this, [Qt 5.14 or higher should be used](https://doc.qt.io/qt-5/highdpi.html) and  [Qt::HighDpiScaleFactorRoundingPolicy](https://doc.qt.io/qt-5/qt.html#HighDpiScaleFactorRoundingPolicy-enum) should be set.

maxnet, 2020-06-03:
>The source of this bug is scaling in Qt. To fix this, Qt 5.14 or higher should be used and 
>Qt::HighDpiScaleFactorRoundingPolicy should be set.

Note that Qt itself does not require scaling factors to be integer (which you can also test by setting QT_SCALE_FACTOR to various values).
By default (HighDpiScaleFactorRoundingPolicy::Unset) it should simply use any fractional value as-is without rounding ( https://github.com/qt/qtbase/blob/60feaae1967aca5c7b1b0f6cb732962ddd1b2115/src/gui/kernel/qhighdpiscaling.cpp#L370 )
So if the scaling factor was 1.5 it could just use that.

Suspect the reason there is no real difference between the 100% and 150% setting is that Qt is taking other factors into account -like monitor DPI- instead of only looking at the Windows percentage.
Not sure what the correct way to deal with this is.

tdewey-rpi, 2023-11-17:
Closing as stale. 

As an aside, I suspect this will only be mitigated by a move to Qt6, which is in our backlog but doesn't have a concrete date as yet.

closed 2023-11-17 by tdewey-rpi

bsimmo, 2023-11-17:
It is still a problem of course, see beta forum.
But hopefully you make the interface nicer and think about accessibility (on Windows) and make use of windows features.
Having to get magnifying glasses out just to read some of the text or stick your head in front of the screen is a bit pants and goes against RPi accessibility statements (certainly Foundation and the research they do)

maxnet, 2023-11-17:
>As an aside, I suspect this will only be mitigated by a move to Qt6, which is in our backlog but doesn't have a concrete date as yet.

What Qt 6 feature would make a difference?

(Imager compiles fine against Qt 6.
We already use that on Mac OS X for M1 support.
Not much advantage of using it on other platforms though, and the library does is significantly larger in file size).

tdewey-rpi, 2023-11-17:
Qt6 apparently carries significant HiDPI rework versus Qt5, especially around multi-monitor support.