raspberrypi/rpi-imager #16: Build fails under CentOS 7.7

tuxmaster5000, 2020-03-06:
The cmake call was happy with all libs, but the compiler can't do it job.
```
LANG=C make -j4
[  5%] Automatic MOC for target rpi-imager
[  5%] Built target rpi-imager_autogen
[ 29%] Building CXX object CMakeFiles/rpi-imager.dir/downloadthread.cpp.o
[ 29%] Building CXX object CMakeFiles/rpi-imager.dir/downloadextractthread.cpp.o
[ 29%] Building CXX object CMakeFiles/rpi-imager.dir/dependencies/mountutils/src/linux/functions.cpp.o
[ 29%] Building CXX object CMakeFiles/rpi-imager.dir/linux/linuxdrivelist.cpp.o
[ 35%] Building CXX object CMakeFiles/rpi-imager.dir/linux/udisks2api.cpp.o
/var/tmp/imagewriter/downloadextractthread.cpp: In member function 'virtual void DownloadExtractThread::extractMultiFileRun()':
/var/tmp/imagewriter/downloadextractthread.cpp:242:43: error: 'ARCHIVE_EXTRACT_SECURE_NOABSOLUTEPATHS' was not declared in this scope
     int r, flags = ARCHIVE_EXTRACT_TIME | ARCHIVE_EXTRACT_SECURE_NOABSOLUTEPATHS
                                           ^
make[2]: *** [CMakeFiles/rpi-imager.dir/downloadextractthread.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /var/tmp/imagewriter/downloadthread.h:17:0,
                 from /var/tmp/imagewriter/downloadthread.cpp:6:
/var/tmp/imagewriter/downloadthread.cpp: In member function 'virtual void DownloadThread::run()':
/var/tmp/imagewriter/downloadthread.cpp:272:26: error: 'CURLOPT_XFERINFOFUNCTION' was not declared in this scope
     curl_easy_setopt(_c, CURLOPT_XFERINFOFUNCTION, &DownloadThread::_curl_xferinfo_callback);
                          ^
make[2]: *** [CMakeFiles/rpi-imager.dir/downloadthread.cpp.o] Error 1
make[1]: *** [CMakeFiles/rpi-imager.dir/all] Error 2
make: *** [all] Error 2

```

tuxmaster5000, 2020-03-06:
It looks like, cmake don't check for the right version of libarchive.

maxnet, 2020-03-06:
Needs recent versions of libcurl and libarchive.
CentOS 7 which was released in 2014, and has the policy of not allowing any real package updates after release (only back-ported security and stability fixes) simply will not do.

If your libarchive version does not support features like ARCHIVE_EXTRACT_SECURE_NOABSOLUTEPATHS a carefully crafted .tar.gz file could overwrite any file on the system you are running the image writer utility on.
So not using that feature is not really an option.

tuxmaster5000, 2020-03-06:
It can be, that the system one it to old, but cmake shout check for the minimal needed version.
Here the code line of the CMake file:
`find_package(LibArchive REQUIRED)`
I don't know which version is the minimal needed one,
Reading the manual of cmake it must be like:
`find_package(LibArchive A.B.C REQUIRED)`
Where A.B.C are the minimal needed version.

commit 6081ca44 (2020-03-09): Linux: test for recent libarchive/libcurl/lsblk

The software requires at least:

- libcurl 7.32.0 (released Aug 12 2013!)
- libarchive 3.2.0 (released Apr 30 2016)
- util-linux 2.27 (released Sep 7 2015)

While technically we could build libcurl and libarchive statically
(like we do for Mac and Windows) if the system has older libs,
it's likely that running the software on systems that old may
give other problems as well (e.g. does lsblk --json (util-linux)
and udisks2 work correctly?), so do not bother.

Closes #16
closed 2020-03-09 by maxnet