raspberrypi/rpi-imager #18: What was the decision to use Qt, rather than something like electron?

stephenwil, 2020-03-06:


maxnet, 2020-03-06:
You can find an electron version of the same utility in the "main" github branch, if you prefer that.

There were a couple small issues with it though.
E.g. the webservers running at downloads.raspberrypi.org have the habit of disconnecting the client if the download stalls for 10 seconds. Which can happen easily if writing to slow SD card storage, and the OS is flushing its write cache to storage.
So you really need to have the ability to use download resumption and reconnect in that case.
Something that was not supported out-of-the-box though.

Digging through 626 MB worth of Javascript module dependencies by different authors, is a bit of a pain.
Sure it is all open source and fixable. But the time needed to locate the piece you want to improve, modify the code, find out the upstream repository of that piece of code is, submit a PR to have the changes incorporated there, is pretty large.
If you have a couple issues like that, and what the application doing isn't too complicated (opening a file and writing it to a block device is not rocket science),  it is faster to rewrite the utility in C++

camccar, 2020-03-06:
It's less resource intensive too! And that is great when running it on a rpi itself.

closed 2020-03-06 by ghollingworth