raspberrypi/rpi-imager #21: Policy regarding third-party operating systems?

dividuum, 2020-03-06:
I wonder if you're open to include third-party operating systems and how that might work in the future. I've built a repository endpoint for [info-beamer hosted OS](https://info-beamer.com/).

You can try it out with:

```
imagingutility --repo https://info-beamer.com/repo.json
```

It can be embedded into an existing configuration using:

```
{
    "name": "info-beamer",
    "description": "Turn your Raspberry Pi into a digital signage display or video wall.",
    "icon": "https://info-beamer.com/repo.png",
    "subitems_url": "https://info-beamer.com/repo.json"
}
```

The images in the repository are automatically up-to-date thanks to the [programmatic OS releases](https://info-beamer.com/doc/api#osreleaseinformation) info-beamer does.

Thoughts?

ghollingworth, 2020-03-06:
I haven't yet decided on our policy here, I'd like to say that it's download driven.  In other words the greatest download numbers will drive the ordering of the list.  But I can't do that because we specifically didn't want to log installation data (since there was another application that did that and they got very bad press for it)

At the same time, this is supposed to be the simplest tool and I don't want to grow the menu too much...

So at the moment I'm torn between adding lots of cool images that show some really interesting uses of the Pi and making it as simple as possible!

Will have a think

dividuum, 2020-03-06:
Just a thought: Maybe a "Third-Party Images (not supported by the Pi Foundation)" submenu would be a solution. In theory that submenu could link to a JSON file managed in this repository as well. That way its content can be manage through GitHub. Not sure how quality management would work though.

procount, 2020-03-07:
I would like it if you add PINN to the list of downloads. I created a "repository endpoint" for it but it won't download PINN until #17 is fixed or I find a new download server location.

ghollingworth, 2020-03-07:
The only issue with a "Third party images" submenu is that Raspberry Pi (actually trading not foundation) only support the Raspbian distribution.  Things like Ubuntu, LibreELEC and OSMC are externally supported by other groups.  But we've had a good relationship with each of them in the past and they clearly provide separate functionality that it is useful to have available.

So it turns out that anything that isn't Raspbian falls into that category.  As I said my personal favourite would be to make this menu a market based on downloads.  So the more people who download (or maybe 'like') an image the more likely it is to appear up the menu.

Maybe we could add a 'like' button for an image, so when people like an image we use these likes to choose the order (again dynamically!)

procount, 2020-03-07:
Here is my JSON fragment to enable PINN, if you'd like to add it.
```
{
        "name": "PINN",
        "description": "A multi-boot OS installer with OS admin features",
        "icon": "https://raw.githubusercontent.com/procount/pinn-os/master/iu/pinn.png",
        "subitems_url": "https://raw.githubusercontent.com/procount/pinn-os/master/iu/os_list_iu_pinn.json"
}
```

bsimmo, 2020-03-08:
(honesty) I've not used NOOBS for a long time.

But I would have thought you would add the list found in NOOBs 'available images to install'* as a minimum.

*Or whatever it is named as.

dividuum, 2021-04-14:
I'd like to bump this issue as I've noticed there is a _digital signage_ mention in the form of _Other specific purpose OS_ section. It would be great if info-beamer could be added as it fits perfectly. Here's the required information to add to the _os_list_imagingutility_v2.json_:

```
{   
    "name": "info-beamer digital signage",
    "description": "Turn your Raspberry Pi into a professional digital signage display or video wall",
    "icon": "https://info-beamer.com/repo.png",
    "subitems_url": "https://info-beamer.com/repo.json"
} 
```

The urls (repo.png and repo.json) are stable urls and is in fact unchanged since I initially opened this issue. And due to how the release process for info-beamer works, repo.json is always up to date. Would it be possible to include this?

closed 2021-04-15 by ghollingworth