raspberrypi/rpi-imager #23: Failure to install Ubuntu Server

DuckyCrayfish, 2020-03-08:
Hello!

Ubuntu Server 18.04.4 LTS fails to install correctly on a Raspberry Pi 2 B v1.1. It fails the validation step and then does not boot.

Raspbian works installs fine on this device, and Ubuntu installs correctly using Etcher.

ghollingworth, 2020-03-08:
This is a known problem that has been fixed, we're just waiting for a couple of other issues before pushing out a new build

ghollingworth, 2020-03-08:
Sorry, will leave it open until the build is pushed...

DuckyCrayfish, 2020-03-08:
Great! Thank you so much

ghollingworth, 2020-03-09:
Fixed and released, please go to the Raspberry Pi Downloads pages to update

closed 2020-03-09 by ghollingworth

lurch, 2020-03-09:
Is there any way for a user to tell what version of rpi-imager they're running?

maxnet, 2020-03-09:
Other then hovering mouse over the .exe (newest version is 1.1 there)
There currently isn't.

lurch, 2020-03-09:
...which I guess are both Windows-specific?