raspberrypi/rpi-imager #24: Calculates drive size differently from other apps

probonopd, 2020-03-08:
Imager calculates the drive size differently from other apps. As a result, it displays a 4 GB microSD card as "3 GB":

[image]

The "Disks" application in Gnome calls it "4 GB", as does the vendor on the packaging:

[image]

maxnet, 2020-03-08:
Some vendors are even selling 15.500.000 bytes cards as 16 GB, because they know certain operating systems round the number anyway.

Should we follow that practice or not?

[image]

What should be displayed is not universal though. E.g. KDE does not round, and displays GiB.

bsimmo, 2020-03-08:
It should use the convention of the OS you are using the program on.  That is what users will expect.

maxnet, 2020-03-08:
So what should the convention of the Linux OS be?
KDE GNOME, some magic detection of the Window manager used? :-)

probonopd, 2020-03-08:
I'd say, typical nominal values of such media are 1, 2, 4, 8, 16, 32, 64, 128, 256, 512 MB/GB/TB. Other values are very, very rare for storage media.

lurch, 2020-03-08:
> Other values are very, very rare for storage media.

I believe it's only SD cards that _always_ have this power-of-2 capacity sizing (give or take whatever size the manufacturer _actually_ puts on the card)? Other flash-based media (e.g. SSDs or USB thumb drives) might also have similar power-of-2 capacities, but this definitely **isn't** the case for HDDs. I wouldn't call hard-drives very rare :wink:

lurch, 2020-03-08:
> It should use the convention of the OS you are using the program on. That is what users will expect.

You could also argue that a cross-platform application ought to be expected to work the same on each OS it runs on? :man_shrugging: :smiling_imp: 

@maxnet Is there a reason for displaying "3,965,190,144 bytes" as "3 GB" rather than "3.9 GB" ?

bsimmo, 2020-03-08:
> Imager calculates the drive size differently from other apps. As a result, it displays a 4 GB microSD card as "3 GB":
> 
> [image]
> 
> The "Disks" application in Gnome calls it "4 GB", as does the vendor on the packaging:
> 
> [image]

What does 
lsblk --bytes 
give ?
It is what the program uses to get the size info (I believe). (with some formatting options and all drives)

probonopd, 2020-03-08:
```
sdb      8:16   1   3965190144  0 disk 
├─sdb1   8:17   1    536870912  0 part /media/me/LIBREELEC
└─sdb2   8:18   1     33554432  0 part /media/me/STORAGE
```

maxnet, 2020-03-08:
> It is what the program uses to get the size info (I believe). (with some formatting options and all drives)

The rounding down to a whole number is my doing, rather than something lsblk did.
Just thought it would be weird to round the number up instead, as that could appear to the user that a 4 GB image could be written to probonopd's slightly smaller than 4 GB card, while that certainly will fail.

Adding a decimal like suggested by lurch could also be a good middle ground.

bsimmo, 2020-03-08:
Decimal place sound, certainly better than loosing almost 1GB good
 or 
use whatever the OS rounds to (so without --bytes)?

commit c04516cb (2020-03-09): SD card selection: round size instead of floor, and display decimal

Closes #24
closed 2020-03-09 by maxnet