raspberrypi/rpi-imager #27: Imager does not save .img files correctly while other program does.

4k3or3et, 2020-03-08:
Hi All

For some reason Raspberry Pi Imager for Windows does not save .img files correctly to SD card. The weird thing is that Win32 Disk Imager saves the .img file properly.

I tried two SD card readers and four SD cards, all the same result.

Basically not matter what .img file I am trying to save using RPi Imager it always fail after verification. Also there are no partitions at all after saving imgae on SD (i was checking in gparted). 

I tried my own backups and official Buster img file downloaded from raspberrypi.org.

When i select from RPi Imager Buster OS which is downloaded from Internet by the program then it saves it on SD card correctly.

So looks like problem lays down in saving .img images.

Anybody can help what is the reason why it does not save  partition table correctly when saving .img files?

Anybody has the same issue?

Thank you.

maxnet, 2020-03-08:
There is a bug in which verification fails if the image size is not dividable by 128 KB.
You should not have a problem writing official Raspbian images which are a nice round number of megabytes large though.

4k3or3et, 2020-03-08:
> There is a bug in which verification fails if the image size is not dividable by 128 KB.
> You should not have a problem writing official Raspbian images which are a nice round number of megabytes large though.

Thank you for your reply. As I wrote before, even writing official buster image leaves SD card unpartitioned...

maxnet, 2020-03-08:
What image exactly?

It is normal for partition table to be empty if validation failed.
First part of card (including partition table) is not written until everything complete. Otherwise there is a large chance Windows will be accessing the card and modifying things while we are still using it.

maxnet, 2020-03-08:
Hmm, thinking about it, the same bug may also occur in a few other circumstances if you are using uncompressed raw images.
Does it work correctly if you feed it an official Raspbian .zip file without uncompressing it first?

If so, then it likely the same bug. Which has already been fixed in git. But did not make it to a new release yet.

4k3or3et, 2020-03-08:
I am taking about this image:

https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/2020-02-13-raspbian-buster.zip

No matter if try to save it from .zip or after decompressing in .img format. Both ways fails for me...

Thanks a lot for digging into my problem. 

Can you tell me please if there and change log i could watch for any chagnes and new version of this tool?

maxnet, 2020-03-08:
>Can you tell me please if there and change log

During development the commit list has all code changes: https://github.com/raspberrypi/imagewriter/commits/
E.g. "Fix verifying on images that are not dividable by block size" is the change that should fix this.

Will leave the question whether there will be a more human readable (aimed at normal users rather than developers) changelog on the site, open for someone else.

ghollingworth, 2020-03-09:
Fixed and released, please go to the Raspberry Pi Downloads pages to update

closed 2020-03-09 by ghollingworth