raspberrypi/rpi-imager #28: Rapsberry Pi Imager makes SD cards unusable

RyanNgCT, 2020-03-09:
My 32GB Samsung EVO micro-SDHC UHS-1 Card cannot be read with my Mac now after failing to be formatted with the Raspberry Pi Imager.

^ I commented this yesterday on the imager page.

**I would like to find out if there is a way to repair the sd card as it is not possible with Mac OS X Mojave's Disk Utility Application.**

Thanks.

Ryan

[image]

Another user had a similar problem.
I quote:
"I have tested this application with 3 different SD cards and in all cases it fails. Moreover after retrying the procedure the SD card become unusable and my Windows 7 computer can not longer read it, it looks like that thy have been damaged by the tool.

I have used 1 32GB and 2 64GB and in all cases the 3 SD cards are broken due to the tool.

Have you experienced this problem?
Any workaround recover my damaged SD cards ?

Thanks,
Antonio"

maxnet, 2020-03-09:
Try the "erase" option in the Raspberry Pi Imager.

RyanNgCT, 2020-03-09:
The problem is the imager does not even detect the card though...
[image]

RyanNgCT, 2020-03-09:
the problem is the card does not even get detected...
[image]

RyanNgCT, 2020-03-09:
Oops accidentally closed the issue...

maxnet, 2020-03-09:
When you insert a SD card without partition table, it should show you a dialog like this:

[image]

Make sure you press "Ignore" there -and not "Eject"-
Then the card do should be visible in the imager.

maxnet, 2020-03-09:
BTW your screenshot shows "Writable: no"
Is it perhaps a full size SD card that has a switch on it to write protect it?

RyanNgCT, 2020-03-09:
Ok it works thanks for your help. But are there still issues with the imager?

maxnet, 2020-03-09:
There should not be a problem with images you download using the utility.
You may have with custom images (already fixed in source)

RyanNgCT, 2020-03-09:
Hmm but can I modify the image after it is flashed? I'm trying to do a headless Raspberry Pi 4 setup.

RyanNgCT, 2020-03-09:
Because I may need to set up the ssh and config files.

maxnet, 2020-03-09:
Just do not use the "use custom" option in the Imager until a new version is released.

RyanNgCT, 2020-03-09:
Ok noted with thanks!

closed 2020-03-09 by maxnet

lurch, 2020-03-14:
> I may need to set up the ssh and config files.

#12 discussed that...

ghollingworth, 2020-03-14:
This issue has now been fixed, you can use the "use-custom" option again...  Just reinstall the application to pull in v1.2

lars-dev, 2021-04-12:
I seem to have the same problem. Two different micro SD cards are broken after use of Raspberry Pi Imager V1.6.1
Both cards are 16GB but different brands.

The first card was never fully able to get an image written on it. It now only shows um as Generic Storage in disk utility. 
[image]

I tried to format the card with disk utility and terminal. It always shows the same error.
[image]

Same issue when trying to erase the card on a windows machine.

When i use the erase option of Raspberry Pi Imager it tells me that it was successful but basically nothing changed.
[image]
[image]

After that when I try to write the image to the card it still shows: 
[image]

this is the info I get in terminal
[image]

When trying to format it with terminal i get the same error:
[image]

This shows up in System Information
[image]

I did a few restarts in between and tried again but nothing. So I assume it is broken. 

The second micro SD card is even weirder. 
It mouted normally and I was able to start writing the image to it with Raspberry Pi Imager. It went up to around 99% (maybe even 100%) and then gave me some error.
I thought, thats odd but since writing worked this time I just wanted to try again.
So i unplugged and the card, plugged it back in..... Nothing
No message about not having the correct file system, no indicator in disk utility, no list in terminal (diskutil list) ....nothing
[image]

The only place I was able to find the card was in the System Information:
[image]

When I try on a windows machine it will show up as /D: .  I am able to change that to  eg /E: but it does not have any partition or similar that I could address to be erased. 
Similar to mac it shows up as a USB Storage but I cant do anything with it.

So that card seems to be (even more) broken too.
The cards were both Class 10.

Any suggestions. I just want to avoid buying and immediately breaking new SD cards.

lurch, 2021-04-12:
Do you have a different SD card reader that it might be worth trying? I dunno if it'll help, but see also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

lars-dev, 2021-04-12:
Yes 
Sorry i did not mention that
Have two different card readers on hand.
Maybe I was just unlucky and the sd cards were close to their last days anyways

lurch, 2021-04-12:
As extensively explained [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) it's not possible for application software to "break" SD cards.

ghollingworth, 2021-04-13:
Were they good quality branded SD cards?  i.e. something like Sandisk?

We currently only recommend Sandisk (we've been testing many different SD cards in an automated SD card test system and the Sandisk cards have been the most reliable so far).

lars-dev, 2021-04-13:
It seems like I just got unlucky. Got new SD Cards and they worked as they were supposed to. The other two cards are still "broken" though. 
And even if that might be the wrong word.... in the end of the day I can not use them anymore. That to me is broken.

mnr, 2021-09-27:
I have two 16gb PNY SD cards that exhibit this same problem. Both Mac, Windows, and RPi believe the disks are write-protected. This happened after reformatting them with the RPi Imager on a MacBook Pro with a USB C Hub Adapter. I'd be happy to mail these two cards to anyone with forensic chops to determine what's up.

I've already tried to remove write-protected files, tried hard-reformat, tried reformat with the SD card formatter from the SD card association, etc.

MNR

Farmerbrown87, 2021-10-06:
Hi, I am having the same issue and have 'bricked' four SD cards. Attempts made with the following bits...

hard
- Raspberry Pi 3 B+ 
- two × SanDisk Ultra 128 GB microSD cards
- two × SanDisk Extreme PRO 64 GB microSD cards (new)
soft 
- Raspberry Pi Imager v1.6.2
- Raspberry Pi OS (32-bit) release 2021-05-07
- Raspberry Pi OS Full (32-bit) release 2021-05-07

@mnr was the issue with the same version of the imager?

Farmerbrown87, 2021-10-07:
@lurch I have tried an old card-reader, Kingston MobileLite G3 (USB 3.0).
The first card I tried was [Anker USB C Hub](https://www.amazon.co.uk/gp/product/B07H4VQ4BZ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). 

I've successfully imaged a diet-pi OS for PiHole using the Anker hub for an OG Raspberry Pi.

I will borrow my partners Macbook and see if that works... watch this space!

Farmerbrown87, 2021-10-12:
Update. 
I have tried writing the microSD card with the Macbook and tried yet another SD card as well.
I have also purchased the official RPi power supply.

Still it is not working. 

What else can I troubleshoot here before I determine the RPi needs to go back to the supplier?

thanks in advance

lurch, 2021-10-12:
@Farmerbrown87 In an earlier message you said that RPi Imager was "bricking" your SD cards, and now you seem to be implying that the problem is only that your RPi isn't booting? Did the verification step in RPi Imager work?
For these kinds of non-booting support issues I think you'll get much more rapid and detailed help on the [forums](https://forums.raspberrypi.com/).

MichaelVanEnde, 2022-05-19:
I have encountered the same isseu with the v1.7.2 release. 
[image]
I tried with 3 SD cards. with differend adapters. Even without changing the settings i encounter the problem.
the restore my SD cards i had to "Erase" it withe the RBP Imager and then i got the following errer:
[image]
after this error you can format the SD card again with via windows.

crstrand, 2022-09-18:
I am having the same issue with Imager 1.7.3
I first noticed it on the windows app (bricked 2 previously good 16GB cards) then with the linux app on ubuntu (bricked 1 32GB card).  All cards were verified good with f3 and the 2 16GB cards had worked in a pair of Pi0W for a few months.
I can't get any of the 16GB cards to be recognised by either my windows machine or my linux box.
I have tried multiple uSD readers and all have the same behavior with the cards.
I DID successfully image on 32GB card on the linux box, then the second card "bricked" with imager popping up a write error message.
I attempted to ressurect the 32GB card with `fsck`, `e2fsck`, `parted` and `mkfs`
However, the second 32GB card now reports 
```
sudo  f3probe --destructive /dev/sdi
F3 probe 8.0
Copyright (C) 2010 Digirati Internet LTDA.
This is free software; see the source for copying conditions.

WARNING: Probing normally takes from a few seconds to 15 minutes, but
         it can take longer. Please be patient.

Bad news: The device '/dev/sdi' is damaged
```
I know some posters have said the app can't brick cards, but the same behavior between two different computers, operating systems, and readers with the only similarity being the Raspberry Pi Imager app is VERY suspicious.