raspberrypi/rpi-imager #29: macOS minimum requirements?  (imager crashes on 10.12.6)

spawnflagger, 2020-03-09:
Does Raspberry Pi Imager for macOS require 10.13 or greater?
(I couldn't find any mention of requirements on raspberrypi.org or the github readme)

I think it's missing a Qt library, or my OS (Sierra, 10.12.6) is too old.

Crash report (sent to apple) excerpt has the following:

```
Dyld Error Message:
  Symbol not found: _OBJC_CLASS_$_MTLCaptureManager
  Referenced from: /Applications/Raspberry Pi Imager.app/Contents/MacOS/../Frameworks/QtGui.framework/Versions/5/QtGui (which was built for Mac OS X 10.13)
  Expected in: /System/Library/Frameworks/Metal.framework/Versions/A/Metal
 in /Applications/Raspberry Pi Imager.app/Contents/MacOS/../Frameworks/QtGui.framework/Versions/5/QtGui
```

maxnet, 2020-03-09:
Yes, Qt expects 10.13 in the latest version.
https://doc.qt.io/qt-5/macos.html

closed 2020-03-09 by maxnet

spawnflagger, 2020-03-09:
thanks for the confirmation.  I just used dd for now, but will try Imager again after I upgrade macOS.  It might be worth adding "_requires 10.13 or newer_" to the download page (or maybe not, since Apple has stoped supporting it).

commit 560759fe (2020-03-09): Mac OS X: disallow being started on 10.12

Qt only supports 10.13/10.14/10.15 in latest Qt version
So mark application as such.

Closes #29