raspberrypi/rpi-imager #30: Ubuntu 18.04 small font on 4K

judgewooden, 2020-03-11:
Ubuntu 18.04.4 LTS
GNOME 3.28.2
Linux mad 4.15.0-88-generic #88-Ubuntu SMP x86_64
Dimensions: 7680x2160 pixels (2032x571 millimeters)
Resulktion: 3840x2160 (16:9)

Seem application is not using Scale percentages. 

[image]
[image]

judgewooden, 2020-03-11:
I noticed a workaround for the same issue on Windows using:
`export QT_SCALE_FACTOR=2
rpi-imager`

This solves the scaling for the main display but get error:
`QObject::setParent: Cannot set parent, new parent is in a different thread`

[image]
[image]

maxnet, 2020-03-11:
Might be specific to window manager used, or some other settings.
Not reproducible under Kubuntu 18.04 (Ubuntu with KDE desktop)

[image]

judgewooden, 2020-03-23:
QObject::setParent: Cannot set parent, new parent is in a different thread

> Might be specific to window manager used, or some other settings.
> Not reproducible under Kubuntu 18.04 (Ubuntu with KDE desktop)
> 
> [image]

I switched to kubuntu and had the same problem with the dropdown. Did a clean install on another machine and did not have the problem with ubuntu 18.04 Gnome or Ubuntu. I played around with resolutions and other options but the problem does not go away for the drop bow.. 

My machine have been a clean install on 16.04 and recently upgraded to 18.04 LTS.

What can I do to help find the cause?

maxnet, 2020-03-23:
>I switched to kubuntu and had the same problem with the dropdown.

Using the same installation, without removing any accessibility settings you had in GNOME first?

My guess you have some special setting somewhere that causes fonts to be drawn larger then the application requests, but does not do any other scaling. (The fonts in your Libreoffice also look disproportionately large to the other controls).

waveform80, 2020-05-16:
I have noticed that Qt-based applications (quassel, keepassx, and the pi imager) don't obey the scaling if it's applied while the application is running, but close and re-open the application and suddenly things are fine(ish). In other words, Qt only appears to read (or apply?) the scaling setting at startup.

If there's an issue specific to the pi imager here, it would be useful to confirm that it's an issue encountered when opening the application fresh after applying scaling settings. Otherwise, this may just be a general upstream Qt issue.

judgewooden, 2020-06-17:
Installed ubuntu 20.04 and the problem went away. I also have a windows 10 machine configured to the same display using HDMI-2 - there I still have the problem.

I have 2x Panasonic TX-40FXW654

akhepcat, 2020-12-18:
FYI -  this is a known issue for all QT apps  on high-DPI   monitors.

setting the environment variable  "QT_AUTO_SCREEN_SCALE_FACTOR=1 "    before starting the app
resolves the issue.

if  rpi-imager  could automatically detect the high-DPI screen and set the scale correctly, that would fix for everybody

judgewooden, 2020-12-19:
okay - you can close the issue

closed 2021-01-17 by maxnet