raspberrypi/rpi-imager #31: Drive Letter Not Released After Verify Sucessful

ZhuDaHai, 2020-03-12:
Windows 10 Pro 64bit.
SDHC Card inserted into Drive F on built-in SD Card Reader of PC.
Local .IMG file written to SDHC Card Via Raspberry Pi Imager v1.2
After Verify is complete and successful, Drive is automatically 'ejected' from the PC, but the Drive letter F remains in Windows explorer and is not accessible.
After re-inserting SDHC Card and manually ejecting SDHC card via the System Tray, then both driver letters F and G are properly removed from Windows Explorer.

lurch, 2020-03-12:
Does the drive letter disappear (after verification) if you remove the SD-card?

ZhuDaHai, 2020-03-13:
> Does the drive letter disappear (after verification) if you remove the SD-card?

Already answered: "After re-inserting SDHC Card..."

lurch, 2020-03-13:
Sorry, it wasn't clear from your previous message exactly what was happening between "After Verify is complete" and "After re-inserting SDHC Card".

ghollingworth, 2020-04-01:
Cannot reproduce on Dell with embedded SD card

Can you check again with the latest v1.2 of the software.  Is there something special about the SD card or the PC you are using?

closed 2020-05-20 by ghollingworth

ghollingworth, 2020-05-29:
https://serverfault.com/questions/11350/is-it-possible-to-mount-unmount-a-physical-hard-drive-in-windows-xp/11368#11368

There might be some useful information given by some of these tools...  On my computer, I've tried opening a file on the disk before running rpi-imager and it still works fine, the drive goes away.  The thing with the file open (happened to be vi) couldn't open the file for saving afterwards as expected...

I'm on Windows Build 18362.778 (Start -> Settings -> System -> About) to get the build number