raspberrypi/rpi-imager #32: MacOS Imager SD Card list misleading

inchworks, 2020-03-12:
On my iMac, Imager shows four other devices, with a blank space underneath, more than large enough to list a fifth device. With the list selected, moving the mouse over the list of devices highlights each of the four shown, with only a thin grey bar beneath to hint that there is a fifth. Only when the mouse is moved to the extreme right of the list (with the list selected), or scrolled, does a scroll bar appear.

Although I am an experienced MacOS (and Windows) user, the first time I tried to use Imager I concluded that my SD card wasn't being detected, and switched to the Windows version for the first installation. The mouse/scrollbar behaviour is standard for MacOS, but normally a list would extend right to the bottom edge of the window with a partial item shown if room, as a cue there are more items.

lurch, 2020-03-12:
Do you have a hi-res display on your Mac? I wonder if this might be related to #13 or #14 or #30 ?

inchworks, 2020-03-12:
It is a hi-res Retina display. Text is the correct size, not perfectly sharp but good enough for the application. It could be something to do with calculating the size of the fifth item, but see the narrow grey bar at the end of the list in this screenshot. That's the fifth item, as highlighted when the cursor moves over it, and clicking it selects the item.

Note that the Operating System list doesn't show the same problem. There, the final visible item in each list or sub-list is cropped.
[image]

inchworks, 2020-03-13:
Looking at #33 I was mistaken to raise this as specific to MacOS. When I ran Imager on my Windows VM there was only one device on the list, so the behaviour wouldn't show.

inchworks, 2020-03-13:
Closed by accident, and reopened.

ghollingworth, 2020-04-01:
Should add something here to show there are more entries in the list, preferably a partial entry...

commit a96450af (2020-05-23): Misc main.qml fixes

- Add support for tooltips
- Show more SD cards in list. Closes #32
closed 2020-05-23 by maxnet