raspberrypi/rpi-imager #33: Windows 10: Doesn't recognize SD card

reibuehl, 2020-03-12:
On my Windows 10 54bit system, the card is not shown in the card selection. Windows sees the card fine as drive J:, but Raspberry Pi Imager only shows me two internal 250 GB SSDs that belong to a dynamic volume / mirrored disk in Windows and what seems to be two empty slots of my SD/CF/MMC/etc. card reader.

maxnet, 2020-03-12:
Can you try if it appears when using scroll wheel of your mouse?
As someone else mentioned the list not spanning the full height, and only seeing 4 items at the time, while there do was a 5th. #32

reibuehl, 2020-03-12:
Yes, scroll wheel works now that you point me to it. There should be a slider or so to show that there are more options and I think that raw devices of dynamic volumes should not be shown.

ghollingworth, 2020-03-17:
Should remove 0GB entries since that's usually how empty MSDs appear in multi-volume readers...

ghollingworth, 2020-04-01:
#48 and #32 

Will close this one

closed 2020-04-01 by ghollingworth