raspberrypi/rpi-imager #35: Don't remove boot/ssh if present in custom image

tim-moody, 2020-03-14:
The following is on windows 10. I can install a custom image, but the ssh file is removed from boot so headless operation is not possible. Please do not do this or add an option to prevent it. The situation is more difficult in that the sd card is unmounted at the end of verify (good), but must be remounted to add the ssh file and then properly unmounted.

lurch, 2020-03-14:
Really? I find it hard to imagine that rpi-imager would contain code to specifically delete the boot/ssh file :confused: 
Are you certain that your custom image contains the boot/ssh file in the first place?

tim-moody, 2020-03-14:
100%

lurch, 2020-03-14:
Can you try writing your image with Win32DiskImager and see if that behaves any differently?

tim-moody, 2020-03-14:
I use dd to write the image and ssh is always there. I create these images on a headless rpi using ssh. I dd them to linux and use dd to write them to another card and am always able to ssh in. 

I mounted the sd card in linux after writing with rpi-imager and the ssh file was gone. The other rpi-setup files that I copy to my images were there.

lurch, 2020-03-14:
> I create these images on a headless rpi using ssh. 

Ahhh, are you trying to `dd` from within a live mounted filesystem? If so, it's possible that the boot/ssh file is only in the disk-buffers in RAM, and hasn't been flushed out to the underlying SD card yet? You can use the `sync` command to ensure that the disk-buffers have been flushed before you run your `dd` command.
Yes, I'm clutching at straws here... :wink:

ghollingworth, 2020-03-14:
This is impossible, we do not delete any files on the SD card...  Check the code if you don't believe us!

closed 2020-03-14 by ghollingworth

tim-moody, 2020-03-15:
My apologies. I hadn't realized that the ssh file is removed on first boot after sshd is enabled and my inability to ssh in resulting from slow startup, probably due to resizing the root partition. In any event I created a new image and things work the same with rpi-imager as with dd.