raspberrypi/rpi-imager #36: Is it necessary to remove country code?

tim-moody, 2020-03-14:
A custom image that had country code set and allowed hostapd to run seems to have had that removed when written with rpi-imager. This prevents locale specific images from being duplicated without the extra work of setting country code again.

lurch, 2020-03-14:
Sounds like a duplicate of #35 ?

tim-moody, 2020-03-14:
apples and oranges are both fruit, but are not the same. probably the same philosophy governs both actions. The behavior is the same as what happens when you create an image on rpi 4 and move it to rpi 3. It always works on 4, but hostapd does not start on the 3.

ghollingworth, 2020-03-14:
Again, we do not do any alterations to the original image...  If you can reproduce this with a single image that when written to an SD card with imager gives you a different output to writing it with dd then I'd like to see the image

closed 2020-03-14 by ghollingworth