raspberrypi/rpi-imager #38: QML warnings on start

alkersan, 2020-03-17:
A few qml warnings are shown in stdout on program start:
```
qrc:/main.qml:739:13: QML QQuickText: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/main.qml:60:17: QML ColumnLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/main.qml:95:17: QML ColumnLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/main.qml:131:17: QML ColumnLayout: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
```

rpi-imager version: [`v1.2`](https://github.com/raspberrypi/rpi-imager/tree/v1.2)
Qt version: `5.14.1`

lurch, 2020-03-17:
Duplicate of #15 ?

alkersan, 2020-03-17:
Yes, didn't find that one

closed 2020-03-17 by alkersan