raspberrypi/rpi-imager #40: Ubuntu OS image options didn't show up and then did show up

dlarue, 2020-03-17:
I'm on Kubuntu 18.04 I built and ran the tool and didn't see the Ubuntu OS options but after installing and running the Raspberry Pi org version(also v1.2) I saw the Ubuntu OS options.  What's strange is after going back to the one I'd built now the Ubuntu OS options show up.

Steps:
Built from source using exactly the commands in the readme. 
I installed using 'dpkg -i' and then opened rpi-imager and didn't see any Ubuntu OS options.
Uninstalled using 'apt remove' and installed rpi-imager from raspberrypi.org
Ran it and the Ubuntu OS option was there.
Uninstalled using 'apt remove', installed rpi-imager_1.2_amd64.deb and SEE Ubuntu OS options.

Is there something I missed during the building and running of the locally compiled version?

ghollingworth, 2020-03-17:
Works as expected, the OS options are defined in a .JSON file on our website, 

http://downloads.raspberrypi.org/os_list_imagingutility.json

When you open the application it populates the list from these which can change dynamically

closed 2020-03-17 by ghollingworth

dlarue, 2020-03-17:
Then what I'm telling you is that it didn't work.

I searched for that file in ~/.local and didn't find it. Where is the os_list_imagingutility.json file downloaded to?

ghollingworth, 2020-03-17:
It isn't downloaded anywhere

dlarue, 2020-03-17:
Well, it didn't work until after I installed the compiled version on the raspberrypi.org site... and then reinstalled the one I compiled.  Why would that be?

XECDesign, 2020-03-17:
I'm guessing that Ubuntu was added to the list around the time you installed the version from raspberrypi.org

dlarue, 2020-03-17:
This happened within a 5 minute period... I had already downloaded the raspberrypi.org version previous to building from source.  I had read this blog and thought I'd try it because it mentioned the Ubuntu OS options so it was already there.

https://www.omgubuntu.co.uk/2020/03/how-to-install-ubuntu-on-raspberry-pi

dlarue, 2020-03-17:
I also had the OS selection window open for a minute or two because I'd never seen this app before and thought I missed something about where the Ubuntu OS options were so I looked into the "misc" option menu item and the "use custom" options.  Surely it shouldn't have taken that long to get the json file.

dlarue, 2020-03-17:
it is doing it right now even...
![Screenshot_20200317_100816]

(https://user-images.githubusercontent.com/3150680/76882061-56fafa00-6837-11ea-9e1c-ac215a117179.png)

dlarue, 2020-03-17:
Just shut down the app and restarted it and now the Ubuntu options are there..
[image]

XECDesign, 2020-03-17:
You're absolutely right. The author of the tool has just pointed out that some of our mirrors are out of sync and were serving up the old json file. You shouldn't be seeing that anymore, and I'll double check that everything is okay now.

dlarue, 2020-03-17:
So within a few minutes the json file at specified in config.h was changing?

 #define OSLIST_URL                        "https://downloads.raspberrypi.org/os_list_imagingutility.json"

I thinking there was a delay in file access and the application didn't wait and continued with just the defaults.

dlarue, 2020-03-17:
This app looks to be populating the menu when it is opened.

maxnet, 2020-03-17:
>So within a few minutes the json file at specified in config.h was changing?
>#define OSLIST_URL "https://downloads.raspberrypi.org/os_list_imagingutility.json"

Note that downloads.raspberrypi.org is not a single server. :-)

```
$ host downloads.raspberrypi.org
downloads.raspberrypi.org is an alias for lb.raspberrypi.org.
lb.raspberrypi.org has address 46.235.231.111
lb.raspberrypi.org has address 93.93.128.211
lb.raspberrypi.org has address 93.93.130.39
lb.raspberrypi.org has address 93.93.130.214
lb.raspberrypi.org has address 46.235.231.145
lb.raspberrypi.org has address 93.93.135.188
lb.raspberrypi.org has address 93.93.128.133
lb.raspberrypi.org has address 46.235.227.11
lb.raspberrypi.org has address 93.93.130.104
lb.raspberrypi.org has address 46.235.231.151
lb.raspberrypi.org has address 93.93.128.230
lb.raspberrypi.org has IPv6 address 2a00:1098:0:80:1000:13:0:7
lb.raspberrypi.org has IPv6 address 2a00:1098:0:80:1000:13:0:8
lb.raspberrypi.org has IPv6 address 2a00:1098:0:80:1000:13:0:6
lb.raspberrypi.org has IPv6 address 2a00:1098:0:82:1000:13:0:7
lb.raspberrypi.org has IPv6 address 2a00:1098:88:26::1:1
lb.raspberrypi.org has IPv6 address 2a00:1098:0:82:1000:13:0:9
lb.raspberrypi.org has IPv6 address 2a00:1098:0:82:1000:13:0:6
lb.raspberrypi.org has IPv6 address 2a00:1098:88:26::1
lb.raspberrypi.org has IPv6 address 2a00:1098:0:82:1000:13:0:5
lb.raspberrypi.org has IPv6 address 2a00:1098:0:80:1000:13:0:5
lb.raspberrypi.org has IPv6 address 2a00:1098:88:26::2:1
```

dlarue, 2020-03-17:
ah, load balancing type stuff, got it.

dlarue, 2020-03-17:
cool, case closed.  Thanks for keeping with me on this.  Everyone wins.

If there's interest in expanding this apps capabilities... I make a script for doing what this app does but after loading the image onto the SD, I mount the partitions and adjust the wifi settings, hostname and locale/timezone.  Very handy for headless setup.
https://github.com/dlarue/Lubuntu-Donkey/blob/master/script-zipdd

Thanks again.