raspberrypi/rpi-imager #41: “Verifying write failed. Contents of SD card is different then what was written to it.”

matiasandina, 2020-03-17:
I'm still able to boot if I cancel the verify process

maxnet, 2020-03-17:
If it's a custom image. Make sure you are running v1.2 from the "releases" page.

matiasandina, 2020-03-17:
I am running v1.2 and it's not a custom image (also, not WiFi for some reason)

maxnet, 2020-03-17:
Tried a different card?

(That it is boots is not proof that the card is 100% good. It will read back the entire card and if any byte is different, will give this error. Can also mean something else is odd on your system, e.g. memory corruption).

matiasandina, 2020-03-17:
Repeated the issue with another SD

lurch, 2020-03-18:
If you want to you could use https://fight-flash-fraud.readthedocs.io/en/latest/ to check that your SD card(s) is/are fully-working at a low-level?
I guess it wouldn't hurt to test with a different SD-reader too, if you have one? :man_shrugging:

wojciechsmigielskijr, 2020-03-22:
I'm getting this same issue using v1.2. :(

ghollingworth, 2020-03-22:
Which image are you trying to write?  Have you tried using dd?  Does that give a different result with the same card?

lurch, 2020-03-23:
Verifying the written data with `dd` is  bit tricky (since it has no built-in verification), but there's some steps [here](https://github.com/raspberrypi/documentation/blob/master/installation/installing-images/linux.md#optional-checking-whether-the-image-was-correctly-written-to-the-sd-card) explaining how to do so.

maxnet, 2020-03-23:
>Verifying the written data with dd is bit tricky (since it has no built-in verification), but there's some 
>steps here explaining how to do so.

Do be aware that you need to skip the first MB when verifying with dd. (Imager does not write the first MB of image, if it errors out).

E.g. to verify a 1500 MB image:

[code]
sudo dd if=/dev/sdX bs=1M skip=1 count=1499 | sha256sum
[/code]

Should give same hash as:

[code]
dd if=uncompressedimage bs=1M skip=1 count=1499 | sha256sum
[/code]

wojciechsmigielskijr, 2020-03-24:
My SD card was buggered. I ordered a new one and the new one works fine.

closed 2020-04-01 by ghollingworth

seongjunChoi0714, 2020-04-04:
I'm getting this same issue.
so 'Balena Etcher' attempt is a success. ('Balena Etcher', the old one.)
There is no problem with SD card.

lurch, 2020-04-04:
@seongjunChoi0714 Would you mind providing more details?
Are you definitely using RPi Imager v1.2? What OS are you running on? What image are you trying to flash? What make/model/size of SD card are you trying to write to? Just for the sake of comparison, what version of Balena Etcher are you using?

seongjunChoi0714, 2020-04-04:
@lurch 
OS : Window10
SD : SanDisk/Ultra/16GB
Imager v1.2
Etcher v1.5.?

lurch, 2020-04-04:
Thanks :+1:  And what image are you trying to flash?

seongjunChoi0714, 2020-04-04:
@lurch  [Ubuntu Preinstalled Server 18.04 image] for ROS2

lurch, 2020-04-06:
I'm afraid I've got no idea what "[Ubuntu Preinstalled Server 18.04 image] for ROS2" means. Do you have a link to a download page?

jeesang7, 2021-04-30:
I want to reopen this issue.
Because getting this same error and then didn't find flashed SD card. But Etcher attempt that was success.
My environment:
- OS: macOS Big Sur
- SD: SanDisk/Ultra/64GB
- Imager: v1.6.1
- Etcher: v1.5.101

sastofficial, 2021-12-04:
this happends to me every time

123sc0ut, 2021-12-12:
Same thing here. Trying to install OpenWRT on a Pi 4B and after 3x memory cards on 3x different card readers on 2x different computers they all come up saying “Verifying write failed. Contents of SD card is different then what was written to it.”
Would love to know a fix

lurch, 2021-12-12:
@123sc0ut Did you buy all 3 SD cards from the same place? Perhaps they're faulty or fake? You can use [H2testw](https://www.softpedia.com/get/System/System-Miscellaneous/H2testw.shtml) to check your cards for errors.

123sc0ut, 2021-12-12:
Ok so I got a work around - rather that using the Raspberry Pi Imager I used Rufus to install the image and it worked perfectly. There was nothing wrong with my cards, looks like the fault was Pi Imager on Windows v1.6.2, not sure why. . .

penguin-number-123, 2022-03-31:
I get the same problem with a brand new 64gb ultra microsd. I'm just trying to put the rasberry pi 64 bit os to it. If I use etcher 1.7.8, with the zip from https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2022-01-28/  , I get flash failed, right after validation. Imager 1.7.2 also fails/

lurch, 2022-03-31:
@penguin-number-123 If both Etcher and RPi Imager are both failing, then perhaps there's a problem with your microSD card or your SD card-reader? (or some security setting on your PC is blocking the low-level access that these tools need?)

rolandog, 2022-04-22:
In my case, I also got the same error:
[image]

Perhaps it has to do with customization?

I had changed the following options:
- Set hostname: `hostname`.local
- Enable SSH
  - Allow public-key authentication only
  - Set `authorized_keys` for `username`:
- Set username and password
- Set locale settings
  - Time zone
  - Keyboard layout
- Play sound when finished
- Eject media when finished
- Disabled telemetry

Leander250, 2022-08-11:
I just fixed it by using an USB Port on the back of my PC instead of the front. I used a Micro-SD-Card to USB adapter.

hosseindamavandi, 2023-06-21:
> I just fixed it by using an USB Port on the back of my PC instead of the front. I used a Micro-SD-Card to USB adapter.

Surprisingly, it worked for me too! It's funny how such a simple solution can resolve technical problems. It's great to hear that you also found success with this workaround!

TheMariday, 2023-09-18:
> I just fixed it by using an USB Port on the back of my PC instead of the front. I used a Micro-SD-Card to USB adapter.

You have got to be kidding me.

Yep. That worked. Why oh why is this a thing?

Leander250, 2023-09-18:
> > I just fixed it by using an USB Port on the back of my PC instead of the front. I used a Micro-SD-Card to USB adapter.
> 
> You have got to be kidding me.
> 
> Yep. That worked. Why oh why is this a thing?
> 
> Is there a fix for this or another issue that we could link that covers this?

I knew that it is recommended to use USB Ports on the back of your PC if you do a BIOS update. The difference is that the ports on the back are directly soldered to the motherboard, while the ports on the front of your PC go through a cable which is only plugged into your motherboard. It kinda makes sense for a BIOS update but felt rather random for this raspberry OS problem.
Another trick is to prefer USB 2.0 ports instead of USB 3.0.

Anyway, thanks for spreading the knowledge. :)

m-o-b-u, 2025-02-24:
I had the same issue with 5/6 SD cards of mine, not willing to give them all up I tried with different versions of rpi-imager from linux, from windows, and even tried with the back USB ports of my PCs with no luck, 
THE BELOW WORKED FOR ME.

1. Copy your Raspberry Pi image onto a USB drive (the image should be in the .img format only, if your image is in .img.gz format uncompress it with- gunzip your_image.img.gz)
2. And with ethernet connection to your powered off RPI board (RPI4B in my case), place the faulty SD card in the SD-card slot of RPI, and connect the USB Flash drive to one of the ports
3. Now immediately after power-on press and hold the shift key until the recovery menu shows up, after sometime it launches RPI-Imager application. You can proceed with flashing the .img like you usually do :)*happy sigh

Hope this helps!

lurch, 2025-02-24:
@m-o-b-u If using the embedded version of Raspberry Pi Imager (which is what you get when holding down SHIFT at power-on) works for all your SD cards, but using Raspberry Pi Imager doesn't work on your PC, then perhaps that means that your SD cards are okay but it's actually your USB SD-card reader which is having problems?

ambiator, 2025-04-27:
I had major issues with this verify problem and then I clicked on the OS sector went all the way to the bottom and clicked on format this is in the pi imager, after this I did the process of OS install, skipped verify and got a message that the OS was installed.

lurch, 2025-04-27:
@ambiator If the verification process of RPi Imager is failing, then that means there's a problem with either your SD card or your car-reader. So be warned that whilst skipping the verification step may _appear_ to work, you may run into OS corruption problems later - i.e. this is entirely at your own risk.

OmerSchechter, 2026-01-17:
For future users, what worked for me: Downloaded the desired image. Used RPi Imager. When it offered to skip verification, I clicked on the button and the Pi boots successfully.