raspberrypi/rpi-imager #44: Don't download cache if it leaves very little space left

mobluse, 2020-03-25:
I think the cache should not be downloaded if you have very little space left. After the cache was downloaded I had only 4 MB left. It might be difficult to reboot after that.

maxnet, 2020-03-25:
On which operating system?

Were you doing anything else while it was writing (e.g. downloading something else)?
It is supposed to check available disk space at the start of the write, and should only cache if there is 5 GB + size-of-image disk space available.

closed 2020-04-01 by ghollingworth

mobluse, 2020-04-07:
It was on Raspbian Buster on Raspberry Pi 4B. No, I was not downloading anything else. It seems this 5 GB + size-of-image check doesn't work.

Botspot, 2020-04-30:
Does anyone know where this tool stores its cache file?

maxnet, 2020-04-30:
Differs by OS.
Typically somewhere under c:\users\<your user name>\AppData\Local or ~/.cache or ~/Library/Caches

qqq89513, 2021-07-12:
On my win-10, 
1. Hitt `win` + `R`
2. Type `%appdata%\..\local\Raspberry Pi\Imager\cache`
3. You'll see what you need

Danrancan, 2023-01-07:
> Differs by OS. Typically somewhere under c:\users\your-username\AppData\Local or ~/.cache or ~/Library/Caches
> 
> AppData folder -like most other system folders- is hidden by default in Windows explorer.

Where are they downloaded on the MacOS version of the app?

maxnet, 2023-01-08:
> > Differs by OS. Typically somewhere under c:\users\your-username\AppData\Local or ~/.cache or ~/Library/Caches
> > 
> > AppData folder -like most other system folders- is hidden by default in Windows explorer.
> 
> Where are they downloaded on the MacOS version of the app?

Did you check the ~/Library/Caches mentioned?

Danrancan, 2023-01-08:
> Did you check the ~/Library/Caches mentioned?

Yes, I couldn't seem to find them there.

maxnet, 2023-01-10:
>>Did you check the ~/Library/Caches mentioned?

>Yes, I couldn't seem to find them there.

Don't have a "Raspberry Pi/Imager" folder there either?
lastdownload.cache itself may or may not exists, as caching do is disabled automatically when you are low on disk space. But the folder do should exists.

[image]

[image]

[image]

[image]

ThePythonDev, 2024-12-21:
What is the path for Ubuntu Linux?

lurch, 2024-12-21:
> What is the path for Ubuntu Linux?

Try looking in `~/.cache/Raspberry\ Pi/Imager/`