raspberrypi/rpi-imager #46: An alarm should sound when it finishes

mobluse, 2020-03-25:
An alarm should sound when it finishes. This way you could do other things and still switch SD card or use it as soon as it is finished.

lurch, 2020-03-26:
I guess that would be useful for some people, but other people might find it annoying - so there would probably need to be an option to turn on or off the alarm?

I believe one of the design goals of Raspberry Pi Imager was to have a simple easy-to-use GUI _without_ any options? :man_shrugging:

ghollingworth, 2020-04-01:
One would assume we can do some kind of OS specific notification which the user can get rid of in their notification settings...

pb66, 2021-01-12:
I agree a ping or something would be useful, we don't all sit and watch the progress bar.

Perhaps a hidden command line option could be added as suggested in issue #144 for a custom desktop shortcat to offer other options to be set?

commit 86f89338 (2021-01-17): Shift+Ctrl+X for advanced users that likes to customize the image

- Adds "hidden" shift+ctrl+X shortcut for eXpert image customization
  options.

Allows one to set certain options on RPI OS images, namely:

* disable overscan
* set hostname
* enable ssh and
  - set Pi user password if using password authentication
  OR
  - set authorized_keys
    (if running Imager on Linux/Mac this will
     have contents of ~/.ssh/id_rsa.pub prefilled)
* configure wifi settings
  (if computer running Imager is connected by wifi it will
  prefill wifi SSID and if on Windows also PSK).
* set time zone and ke [... cut]
closed 2021-01-17 by maxnet