raspberrypi/rpi-imager #53: rpi-imager destroyed the microSD card, I believe

mobluse, 2020-04-10:
I was trying to flash Ubuntu 19.10.1 64-bit server (built-in to Imager) on to a NOOBS card that came with a Raspberry Pi 4 Desktop Kit (SanDisk SDHC 16 GB). Then it said it couldn't verify. I then tried to erase the card, but that didn't work. I then tried to erase it with fdisk and create a partition, but that didn't work, since /dev/sda1 doesn't get created I couldn't format it. I did all this on Raspberry Pi 4 B 4 GB using a USB 2.0 microSD card reader. I have used this NOOBS card since summer 2019. It is strange if this card would break just when I write an image to it, so I believe the card was destroyed by rpi-imager. I used Imager to create the current Raspbian I use from an empty 32 GB card using the same USB 2.0 microSD card reader. I use Imager 1.2 in most updated Raspbian Buster.

I was running boinc-client with Einstein@home when I flashed the first image, and the boinc-client used two cores. I believe that flashing to a NOOBS and running BOINC at the same time was to complicated.

Then I tried to use SD Card Formatter 5.1 on a Windows 10 computer, but that formatting fails also.

I also tried another microSD card reader in Windows 10 (the built-in), but no change. I now also installed Imager for Windows 10, but with the same result with this SanDisk card.

I included debug.txt from SD Card Formatter for both quick and overwrite format.
[debug1.txt](https://github.com/raspberrypi/rpi-imager/files/4459305/debug1.txt)
[debug2.txt](https://github.com/raspberrypi/rpi-imager/files/4459306/debug2.txt)

lurch, 2020-04-10:
> I believe the card was destroyed by rpi-imager.

AFAIK, that's not actually possible. Sometimes the partition-table on SD-cards "[confuses](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos)" normal partitioning tools - see if any of the tips [here](https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives) help?
If none of those work, `sudo dd if=/dev/zero of=/dev/sdX` (with the X replaced with a suitable letter - use with caution!) should sort things out?

mobluse, 2020-04-12:
`sudo dd if=/dev/zero of=/dev/sdX` didn't help in my case, because the card has probably been write protected for ever. (It took very long time to run, but I run it to the end, and checked progress using pkill.) It doesn't complain when one writes to it, but nothing is written. One might be able to figure out which program that was the last used before write protection happened by studying an image of the card. There are no NOOBS partitions (or any other partitions) on it so the destruction probably happened during the early writing of the card using Imager to write Ubuntu 19.04 64-bit. One theory is that the NOOBS partitions are difficult to remove and that this leads to a lot of writes that destroys the card.

closed 2020-05-18 by ghollingworth