raspberrypi/rpi-imager #55: Windows Support

JinxDojo, 2020-04-10:
Regarding: https://www.raspberrypi.org/downloads/

The link to the imager has no information about which operating systems it supports. Is Windows XP still supported? Personally, I receive an error when launching "rpi-imager.exe": "Entry Point Not Found \n The procedure entry point CancelIoEx could not be located in the dynamic link library KERNEL32.dll".

This could be due to a need to update Qt--or perhaps my operating system is simply no longer compatible. In either case, it would be nice if it stated this somewhere since the installation otherwise went smoothly.

I don't need help personally (I wound up successfully writing the Raspbian image using Rufus v2.18 which was the last version to support Windows XP), but I'd like to save others the time/frustration of trying methods that won't work or being uncertain how to proceed (I've had to bug the folks on IRC more than a couple times).

mbernardi1961, 2020-04-26:
[image]
I tried to install this on Windows 7 and when I try to start it I get the attached error.
FWIW I successfully downloaded Raspberry Pi Imager from the Ubuntu repository

ghollingworth, 2020-05-20:
@XECDesign maybe just add the list of Windows OS support to the downloads page?

@maxnet Do you have the minimum requirements?  I'm guessing it requires Windows 10 to run?

maxnet, 2020-05-20:
>@maxnet Do you have the minimum requirements? I'm guessing it requires Windows 10 to run?

The requirement is that you have a graphic card with some form of hardware acceleration.
Which Microsoft listed as a hardware requirement since Windows Vista.
So technically this error should not occur if you meet OS vendor requirements.
Think XECDesign mentioned seeing the same error when running under virtualizated environiments though. So I guess those do not have that.

Other than that we require at least Windows 7 because of some power management function called (to prevent laptops going to power savings sleep mode while writing).

XECDesign, 2020-05-20:
Yeah, I saw that on a Windows 7 install running in VirtualBox. @mbernardi1961, were you running Win7 in a VM or not?

andrum99, 2020-05-24:
Ideally the installer should display a warning if the user attempt to install on a version of Windows before Windows 7, or just refuse to install. The minimum OS requirements should be listed somewhere on the website as well though - probably on a separate page, as to list all the requirements for each platform is probably going to be quite a lot of information.

I think expecting new software released in 2020 to work on Windows XP is a bit unrealistic - it's been EOL for six years.

closed 2020-05-24 by maxnet

andrum99, 2020-05-25:
Just tried this in a Windows 7 Pro VM under VMWare Workstation on a Windows 10 Pro host. With "Accelerate 3D graphics" enabled (which is the default) Imager works fine. If I disable this option, I get the error about OpenGL show above.

A quick Google reveals that Windows only has software emulation for OpenGL 1.1 - it needs hardware acceleration for 2.0 and higher. Imager should probably bail out more gracefully, but a note on the "system requirements" page would also be useful.

maxnet, 2020-05-25:
>A quick Google reveals that Windows only has software emulation for OpenGL 1.1 - it needs 
>hardware acceleration for 2.0 and higher.

With the configuration we use, it is supposed to use DirectX instead of desktop OpenGL.

I suspect that DirectX may have better software emulation support in later Windows versions. As Windows 10 does work under Virtualbox.

andrum99, 2020-05-25:
You're right - Windows 8.0 onwards seems to have ANGLE, which provides software-emulated OpenGLES 2.0 as a minimum.

maxnet, 2020-05-25:
>You're right - Windows 8.0 onwards seems to have ANGLE

We also bundle ANGLE. It is the libEGL.dll/libGLESv2.dll/D3DCompiler.dll in the imager direcotry.
Still that seems to require that the GPU at least accelerates a little bit on Windows 7 though.
While that does not seems to be the case on later Windows versions.

Not sure if this is a problem that exists for real users on real hardware though.
As said, it has been a hardware requirement by the OS vendor for a long time. And if people do not have drivers for their GPU that is often very noticable in other ways as well. 640x480 graphics, etc.

andrum99, 2020-05-25:
> We also bundle ANGLE. It is the libEGL.dll/libGLESv2.dll/D3DCompiler.dll in the imager direcotry.
> Still that seems to require that the GPU at least accelerates a little bit on Windows 7 though. [...]

Agreed. I don't think this really matters for Imager, but it should probably be mentioned somewhere in the system requirements since someone has already come across it.

maxnet, 2020-05-25:
>I successfully downloaded Raspberry Pi Imager from the Ubuntu repository (but doesn't want to write to the SD card;-().

What Ubuntu repository exactly has an Imager that does not work?

Did you download the .deb from the Raspberry Pi website, or is this an Appstream or other third-party packaged thing?

JinxDojo, 2020-05-25:
> I think expecting new software released in 2020 to work on Windows XP is a bit unrealistic - it's been EOL for six years.

I, of course, did not "expect" it to work, but I did expect it to warn me somewhere (either in the download page or the installer) of the reasons it would not. I'm glad you agree that it'd be helpful to list the minimum requirements on the website (perhaps a side link to "(Requirements)" would avoid cluttering the download page). That would be plenty sufficient and save updating the code.

(On a side note: to be honest, it doesn't seem like an image-writer should require such complex software/hardware [a graphics card with hardware acceleration??] as to render legacy OSes obsolete, but I fully admit that my lack of experience in maintaining software could be biasing my opinion. Fortunately, as I mentioned in the original post, and want to reiterate here to help anyone struggling: Rufus v2.18 still works to write the image using Windows XP.)

Thanks to the community for looking into this issue.

maxnet, 2020-05-25:
>On a side note: to be honest, it doesn't seem like an image-writer should require such complex 
>software/hardware [a graphics card with hardware acceleration??] as to render legacy OSes obsolete

It is not the graphics acceleration that renders legacy operating systems obsolete.
You can add another 15 MB library to the mix to have that done in software, if that was the only problem.

The problem is that we use third-party components, and none of those support XP anymore.
And using older versions of those libraries is not always an option.
E.g. latest version of OpenSSL no longer works on XP. But if you use old version it may not be able to connect to webservers only speaking newer TLS protocols, etc.

XECDesign, 2020-05-25:
It may also be worth noting that Rufus doesn't seem to support Linux or MacOS. If you don't want to write three different native frontends, you use a toolkit like GTK or, in this case, Qt. As a result, you end up dragging in their system requirements. Cross-platform UIs are still a bit tricky so you end up with things like Electron.

Thermoflux, 2021-02-10:
I had the same error when trying to use the imager on Win 10 Pro. 
I resolved it by disabling my Nvidia GPU and only using the Intel GPU.

commit f3bc47a3 (2020-05-24): Windows installer: refuse installation on ancient versions

Closes #55