raspberrypi/rpi-imager #56: Add feature to make an image from an SD card without empty space

D4rkWrench, 2020-04-17:
Having the feature to make an iso from an SD card including ONLY actual content (without any empty space from the SD card) so that making backup becomes way easier than using dd and PiShrink.
Also make this option compatible with windows.

ghollingworth, 2020-04-17:
The Imager tool was not developed for this purpose, it is not possible to correctly copy ext4 fileysystems on a Windows computer without running a virtual machine.

closed 2020-04-17 by ghollingworth

lurch, 2020-04-17:
@D4rkWrench Does the SD Card Copier tool in Raspbian help you do what you want? https://www.raspberrypi.org/blog/another-update-raspbian/

D4rkWrench, 2020-04-21:
It could have worked if the SD Card Copier was able to output an image and not directly to another card...

lurch, 2020-04-21:
ping @spl237 as the SD Card Copier is his baby :baby_symbol: :slightly_smiling_face:

spl237, 2020-04-21:
Not something I have the time to do at present, and completely different functionality from what the tool already does. It'd be nice to have, but it would involve writing a completely new tool, to all intents and purposes; the only bit you could reuse would be the GUI!

D4rkWrench, 2020-04-21:
Maybe it'll be possible to use some parts of pi-Shrink ? I'm not a dev but if it is possible to not make a 16gb file (for a 16gb card) before shrinking it and just reading the existing "true" content from the partitions that could be really useful :)