raspberrypi/rpi-imager #57: Noninteractive command-line interface

fenhl, 2020-04-21:
It would be nice to have a CLI version of this tool where I can supply the operating system and SD card to use as arguments and have the process run automatically without requiring user input.

lurch, 2020-04-21:
Doesn't `dd` cover that use-case? https://www.raspberrypi.org/documentation/installation/installing-images/README.md

fenhl, 2020-04-21:
I tried to use `dd` but it's failing with an error “dd: /dev/rdisk2: Input/output error” after a while. This imager works though.

lurch, 2020-04-22:
Maybe someone on the [forums](https://www.raspberrypi.org/forums/) will be able to help you with your `dd` problem?

fenhl, 2020-04-22:
It turns out that having Python pass the image to `dd`'s stdin doesn't work for some reason and saving it to disk first is a workaround. However it would still be nice to be able to use something that stays up to date with the recommended way of imaging Raspbian and any associated backend changes that might happen in the future.

ghollingworth, 2020-05-20:
This is a bit beyond the rpi-imager tool, especially since the tool needs to refer to the right path to the SD card, which is quite significantly different in structure on the different operating systems.  Also you need to check you're not accidentally overwriting the hard drive, and check with the user "Are you sure (y/N)"

closed 2020-05-20 by ghollingworth