raspberrypi/rpi-imager #59: Empty dialog shown if SD card is read-only

Sleekwater, 2020-05-01:
Windows 10, Imager v1.2

I've got an adapter from SD to microSD that happens to have a "read only" switch. Which, inevitably, was on.
If this switch is set then when you try to find the SD card, Imager will
* Take ages
* Give you a blank dialog if/when it does finish

Check that the SD card is writeable, and provide a suitable error message if not.

maxnet, 2020-05-01:
Not sure if it is possible to find out if a device is read-only in all cases.
It is already supposed to hide devices that are reported by the OS to be read-only in the SD card selection screen (although that was added to hide a number of Mac OS X system devices, instead of read-only cards).

RobCranfill, 2020-06-03:
Perhaps showing read-only devices as present but disabled for use, like the list item being grayed out and disabled, would be more useful. Even better would be something like a little "lock" icon next to it.

_Please accept this as a humble suggestion from a non C++-programmer. I'm surprised how many simple suggestions here are greeted with rancor - "If you don't like the program, fix it yourself!" - when I fully accept that much wonderful work has been done here, and I'm simply trying to assist in some small way in making it better, and to help others avoid a problem that had me puzzled for several hours._

ghollingworth, 2020-06-03:
Unfortunately, as @maxnet has pointed out there is no consistent way of telling whether the drive is read only or not at the moment.  We'll have a look the next time we go through an update cycle, but currently it doesn't look like it'll work properly on all OSes

commit 39abeb23 (2020-07-03): Improve drive listing

- Show write protected SD cards greyed out.
  Show message if user select one of those. Closes #59
- Do show virtual drive types on Windows.
  Do keep them hidden on Mac OS X as that seems to
  cover .dmg file and timemachine stuff. Closes #79
closed 2020-07-03 by maxnet