raspberrypi/rpi-imager #60: Doesn't redraw if window moved

kaihendry, 2020-05-08:
[image]

I'm using Arch btw with the current package of https://aur.archlinux.org/packages/rpi-imager/

lurch, 2020-05-10:
Are you using a non-standard window-manager?

kaihendry, 2020-05-13:
https://en.wikipedia.org/wiki/Dwm ... the standard tiling one

lurch, 2020-05-13:
I've never used a tiling wm myself, but perhaps the GUI used by RPi Imager isn't "compatible" with tiling window managers? :man_shrugging: 
(I should also clarify that I'm not the RPi Imager developer, so I apologise for any incorrect comments / guesses!)

tdewey-rpi, 2023-11-17:
Closing as stale. Please raise a new issue if you see this wiht the current release.

closed 2023-11-17 by tdewey-rpi