raspberrypi/rpi-imager #61: Windows: Not going to use this virus-style program

bmaehr, 2020-05-10:
I'm not going to do anything with a program (especially a program doing something with drives/hard disks), that first requests administrator rights and in the next step only offers an "Install" button and telling me nebulously somthing about installing and upgrading system components. 

Why does an imager need some kind of installation? Why is it not enough to unzip a zip archive if not an exe without installation is enough?

I definitifly want to know what kind of things a program, that I'm going to use mostlikly once, is going to install on my system and I think I would refuse the installation of some frameworks (perhaps Qt or similar) in most cases.

ghollingworth, 2020-05-10:
You realise you're asking this in the gitub repository that contains the whole code inside that installer?  Go and have a look if you don't trust us.

closed 2020-05-10 by ghollingworth

calston, 2020-05-28:
As a better answer any disk imaging tool requires access to the underlying disk hardware to do its job. As such administrative access is required, unfortunately Windows doesn’t provide granular permissions to different system calls

maxnet, 2020-05-28:
Think the reported is not worried about the admin access, but the standard text from the NSIS installer that system components may be updated.
While that is not actually the case.