raspberrypi/rpi-imager #65: Empty error dialog shown if .img is on a network share

ITaluone, 2020-05-20:
Win10 x64
v1.2

It is not possible to write an image which is on a network share. With "W32 Disk Imager" this is not a problem at all.

Reproduce:

- Select "Custom Image" from e.g. \\fileserver\share\to\custom.img (btw. i have rw acces to this share)

- Select SD card

- Try to write the image --> fails

andrum99, 2020-05-20:
Do you mean UNC paths e.g. \\machinename\share\imagename.img? That case should be handled, but it's fairly uncommon for folk to do that, I would have thought. I think for users that know about UNC paths, they probably also know that they can map a drive to the share and use that instead, or copy the file to a local drive.

ITaluone, 2020-05-20:
Yeah of course I know to map the share to a drive or to copy the file to a local drive. But this is what i have tried to avoid. 

Anyway, it should not be an empty dialog.

ghollingworth, 2020-05-20:
Yeah, I have reproduced this at home trying to write an image from my local server through windows sharing...

lurch, 2020-05-20:
This is an issue that Etcher also has, e.g. https://github.com/balena-io/etcher/issues/3122

In https://github.com/balena-io/etcher/issues/1436#issuecomment-301723169 and https://github.com/balena-io/etcher/issues/2858#issuecomment-513587448 I explained what the problem is. Although those explanations don't explain why Win32DiskImager _does_ work with images on a network drive...? :confused:

maxnet, 2020-05-23:
>Reproduce:
>Select "Custom Image" from e.g. \\fileserver\share\to\custom.img (btw. i have rw acces to this share)

How exactly?

I cannot even enter an UNC path on my Windows 10 enterprise VM.
Can only select from dropdown in the location box, and if I just enter an UNC path in the "filename" field imager refuses to accept the file selection, saying "Item selected is not a regular file".

[image]

andrum99, 2020-05-23:
I have been able to reproduce this on Windows 10 Pro (not in a VM) as follows:

1. In file picker dialog, enter the UNC path for the server only, then press enter. For example `\\\\raspberrypi`.
1. The file picker dialog displays the available shares on the server. Navigate to the share, choose the file, then click the 'open' button.
1. Choose the destination device.
1. Press the 'write' button.
1. A blank error dialog then appears.

Screenshot:
[image]

I've also tried typing the server name and share, i.e. `\\\\server\\share`, as well as `\\\\server\\share\\directory` and it shows the same problem. I'm guessing as long as you don't type in the actual filename it will allow you to then go on to choosing the file in the box above.

andrum99, 2020-05-23:
> This is an issue that Etcher also has, e.g. [balena-io/etcher#3122](https://github.com/balena-io/etcher/issues/3122)
> 
> In [balena-io/etcher#1436 (comment)](https://github.com/balena-io/etcher/issues/1436#issuecomment-301723169) and [balena-io/etcher#2858 (comment)](https://github.com/balena-io/etcher/issues/2858#issuecomment-513587448) I explained what the problem is. Although those explanations don't explain why Win32DiskImager _does_ work with images on a network drive...? 😕

Those bugs in Etcher appear to be unrelated. One is Linux, the other is about problems writing images stored on mapped drives. Raspberry Pi Imager on Windows works fine if the network share is accessed using a mapped drive. The problem in this bug report is with UNC paths.

lurch, 2020-05-23:
Maybe I'm simply misunderstanding the problem then - I thought that the problem was that the 'GUI part' of the program (either RPi Imager or Etcher) was running as the "regular user", and it was only the "regular user" who had the credentials to connect to the network share. And thus the "regular user" can browse the network share and select the image to flash, but when the 'flashing part' of the program switches to the "elevated user" (in order to be able to perform raw block-level writes to the disk), the "elevated user" _doesn't_ have the credentials to access the network share, and thus is unable to read the image from the network share.
Based on that understanding, I believe that Etcher and RPi-Imager would be equally affected by the problem on either Windows or Linux...

> Raspberry Pi Imager on Windows works fine if the network share is accessed using a mapped drive.

... so it looks like perhaps I was wrong then, in which case sorry for the noise!
Although I wonder if the behaviour is different depending on whether the network share allows anonymous access, or requires a password-login? :shrug:

maxnet, 2020-05-23:
Mapped drive behavior may also depends on user's security settings, see: https://support.microsoft.com/en-us/help/3035277/mapped-drives-are-not-available-from-an-elevated-prompt-when-uac-is-co

But indeed think this may be a different issue.

andrum99, 2020-05-23:
On my system, Raspberry Pi Imager elevates privileges via UAC when it starts up, so the whole thing is running with admin rights, as I understand it.

maxnet, 2020-05-23:
>In file picker dialog, enter the UNC path for the server only, then press enter. For example \\raspberrypi

At first that did not work for me either.
But seems that was caused by that I did not have the password to access the file share stored inside Windows.
If I access the share first through Explorer and choose to save password there, I do indeed can go to share in Imager, and reproduce the problem.

Now as far as the actual problem goes.
Imager uses libcurl to download files, and also uses that to read "local" files to have same code flow.
We pass the name of the file as URL to libcurl.
I have a strong suspicion that Qt and Libcurl are not on the same page on what an URL should look like when it refers to an UNC path.

commit 71eefa47 (2020-05-23): Fix handling of UNC file URLs

- Qt thinks UNC URLs should look like: file://1.2.3.4/share/file.img
libcurl likes them: file:////1.2.3.4/share/file.img

So correct that.

- Mention it is an unspecified libcurl error if libcurl passes us
an error code with an empty error message string.

Closes #65
closed 2020-05-23 by maxnet