raspberrypi/rpi-imager #66: SVG icon

ghisvail, 2020-05-20:
Right now, the [icon](https://github.com/raspberrypi/rpi-imager/blob/qml/icons/rpi-imager.png) is a fixed-size PNG file, which is acceptable by today's Linux packaging standards.

It would be nice to provide a SVG version of the icon for desktop environments which support dynamic scaling.

lurch, 2020-05-20:
@spl237 do you have a suitable SVG?

spl237, 2020-05-20:
No, I don't use SVG icons anywhere, as they're a pain to create. Might be worth asking the comms team or design if they have one.

ghollingworth, 2020-05-20:
@jackbenwillis

spl237, 2020-05-20:
That said, there probably isn't a huge benefit of going to all the hassle of creating an SVG as opposed to just having a few bitmaps at appropriate sizes; unless Jack is starting from vectors anyway...

jackbenwillis, 2020-05-20:
I generally prefer to work in vector. And only have to make the design once. Although for the new icons Simon and I worked on I was working mostly in bitmap since you get some weird approximations at smaller sizes, which you can remedy by stripping out detail at pixel level.

Which icon do we need a SVG for?

headblockhead, 2024-10-19:
Sorry to revive an old issue, but I designed my own 'unofficial' SVG version using the raspberry pi logo from raspberrypi.com if anyone wants to use it - see #949 for the lack of my own pull request. 

If anyone, including Raspberry Pi's design team, want to use this file, I give full permission.

(the low-resolution PNG icon really bugs me next to all the other vector icons.)

[new file](https://raw.githubusercontent.com/headblockhead/rpi-imager/refs/heads/qml/rpi-imager.svg) 
[image]
[old file](https://raw.githubusercontent.com/raspberrypi/rpi-imager/refs/heads/qml/debian/rpi-imager.png)
[image]
comparison:
[image]

ghisvail, 2024-10-28:
That's very nice, looks amazing

PortalWalkerCodes, 2025-04-22:
Looks amazing!

nafanz, 2025-04-22:
Yeah, it looks good.

tdewey-rpi, 2025-12-12:
SVG icons are present from 2.0.2.

closed 2025-12-12 by tdewey-rpi