raspberrypi/rpi-imager #67: [request] Noobs in list

ale-rinaldi, 2020-05-20:
Hello,

I just tried RPI Imager and I loved how it's fast to write and also to verify the SD card! The fact that it flashes why it downloads it really awesome, congratulations!

I wonder if it could make sense to have Noobs in the list of flashable images, in case one wants to flash it on a bunch of SD cards and decide later which image to actually use on each one.

Thanks

ghollingworth, 2020-11-09:
Just download the image and hit custom

closed 2020-11-09 by ghollingworth

ale-rinaldi, 2020-11-09:
Well yes, I got that. I think it could be useful to have it listed with the other images, but I'll take it as a no.

Thanks

lurch, 2020-11-09:
Problem is that NOOBS isn't a "raw disk mage" in the same format that Raspberry Pi OS (etc.) is. With Raspberry Pi OS, you just write the raw image directly to the SD card (and the image contains the partitions, filesystems, etc.); with NOOBS you have to format the card as FAT32 first, and then copy the files from inside the NOOBS zipfile onto the FAT32 filesystem.

ale-rinaldi, 2020-11-09:
Ok, that totally makes sense. Thanks for your clarification!

guysoft, 2020-12-02:
It seems now that if you install the noobs zip file the imager formats the disk and installs it accordingly.

lurch, 2020-12-03:
See also the comment here: https://github.com/raspberrypi/noobs/issues/610#issuecomment-729041932