raspberrypi/rpi-imager #68: [Feature Request] Preinstallation of SSH Keys

Adam-Langley, 2020-05-21:
One of the first things that I do after imaging and launching my OS, is to install SSH keys for password-less SSH.
It would be a great help, to have a checkbox that enables SSH, and generates/installs the SSH keys into the disk image.

https://www.raspberrypi.org/documentation/remote-access/ssh/passwordless.md

lurch, 2020-05-22:
Duplicate of #12

ghollingworth, 2020-05-27:
The Raspberry Pi Imager application was designed with simplicity in mind, giving the user the smallest number of choices, in the simplest language possible.

Any request that requires extra 'options' or buttons, or may in any way confuse a beginner will not be implemented.

This functionality could only be implemented by adding extra options to the user interface.

closed 2020-05-27 by ghollingworth