raspberrypi/rpi-imager #71: Windows 10: Cannot open/launch

JRHeaton, 2020-05-26:
Windows version: Microsoft Windows [Version 10.0.18363.836]

Problem: When I open Raspberry Pi Imager, no window appears. It asks me for permission, which I grant, then nothing. If I hover over the task bar preview, it shows a blank white preview but no white window is actually accessible in the window space.

Steps taken to resolve: Uninstall and reinstall, separated by a reboot.

Notes: It did work before, and now it does not. Is there perhaps something that could have gotten corrupt, a cache or preferences area that the app is stuck on trying to load? Nonetheless, this is preventing me from using the utility on the latest Windows 10.

ghollingworth, 2020-05-26:
I had a similar problem to this which only occurs when I've got my laptop lid closed (I normally use a secondary external screen.  It would only appear in the taskbar as a white box but didn't appear on the screen.

I then pressed Alt-Space to bring up the window dropdown menu (it always appears on the screen even if the window is off-screen) then you select 'Move' and press the left or right arrow key.  Then move the mouse and it should drag the window back onto the main display...

JRHeaton, 2020-05-26:
@ghollingworth that fixed it. Thanks!

closed 2020-05-26 by JRHeaton

kekbait, 2020-06-30:
@ghollingworth 's solution didn't worked for me,
because i'm working on a multiple-screen setup.

The secondary screen I once used was missing,
and I didn't have it anymore.

I had to open the Register, and delete the following keys :

HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager\General\x
HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager\General\y

Then, it worked perfectly again!

lurch, 2020-07-22:
Sounds like this has now been "fixed properly" #91
(although only in the source-code, not yet in a published version of rpi-imager)

rusagent, 2020-09-16:
This has not been fixed. I downloaded the new 1.4 from the website and the window was somewhere completely offscreen.

maxnet, 2020-09-16:
>This has not been fixed. I downloaded the new 1.4 from the website and the window was somewhere completely offscreen.

Has been fixed in the sense that the source code on github has the fix.
No new binary releases have been made since though.

hamzafarooq951, 2021-01-29:
> @ghollingworth 's solution didn't worked for me,
> because i work on a multiple-screen setup.
> 
> The secondary screen I once used was missing,

> and I didn't have it anymore.
> 
> I had to open the Register, and delete the following keys :
> 
> HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager\General\x
> HKEY_CURRENT_USER\SOFTWARE\Raspberry Pi\Imager\General\y
> 
> Then, it worked perfectly again!

@kekbait Thanks for the solution
I installed the latest 1.5 version and this issue is still there

maxnet, 2021-01-29:
>I installed the latest 1.5 version and this issue is still there

On Windows or something else?
External monitor was disconnected completely (not just turned off)?

ComputerDoktor, 2021-05-01:
Please reopen Issue and fix the problem still existing in 1.6.1. @gsh I would like to remind you of your goal 
trying to make it as easy as possible to get started.

ghollingworth, 2021-05-04:
Still waiting for someone to describe a reproducible scenario, since we can't reproduce it we can't find out what it is and therefore fix it.

The original issue described here was fixed and so the bug is closed.  If you've got a reproducible scenario we can use to reproduce then we can reopen it.

ComputerDoktor, 2021-05-14:
additional fix in imager-1.6.2, see https://github.com/raspberrypi/rpi-imager/issues/220

lipflip, 2021-09-23:
This issue is not resolved. Just installed 1.6.2, the imager won't launch and I don't see an icon in taskbar.

I only have one screen.