raspberrypi/rpi-imager #72: No Verify

procount, 2020-05-26:
Would it be possible to add a `--noverify` option for the more advanced user? It would save some time when trying to flash images in succession, for those who know the risks.

ghollingworth, 2020-05-27:
Probably would be best as a pull request (I'm unlikely to approve this functionality as a general addition, but pulling in a change and give it the once over sounds simple enough)

procount, 2026-01-22:
I'm still using the old version. :wink:
But I looked at the documentation and it looks like it would do the trick.
Is the --cli option still available? If so, then, a '--skip' or '--noverify' option would be useful to set this feature for automating the image writing process in a script or whatever. 
Otherwise, you can close this as complete.