raspberrypi/rpi-imager #79: No entries in the SD Card list

RadianM, 2020-06-07:
Windows 10, Lenovo laptop with built-in card reader: "choose sd card" dialogue has nothing in it even though card inserted and visible to OS. I checked the write lock is off on the card adapter, ejected card, re-inserted, rebooted etc. no luck. Any ideas?
[image]

maxnet, 2020-06-07:
Does the previous version ( https://github.com/raspberrypi/rpi-imager/releases/tag/v1.2 ) work any better?

(Just wondering if the change to filter out zero sized drives and such is causing this).

RadianM, 2020-06-07:
Er, well, yes that version does show the card but now I get "Error writing file to disk". Will try a few more things. But yes, version 1.3 was not showing the card whereas version 1.2 does.

RadianM, 2020-06-07:
Formatting fails too:
[image]

maxnet, 2020-06-07:
The error message during formatting is truncated (failed to allow extended DASD is not fatal)
If you open a "command prompt" and do:

```
cd \program files (x86)\raspberry pi imager
fat32format E:     (double check drive letter first)
```

Do you get an error message at the bottom?

Also if you do:

```
diskpart
list disk
```

Look at the disk number next to the SD card, and select that.
E.g.:

```
select disk 2
detail disk
```

Does it show anything out of the ordinary. E.g. a read-only state?

RadianM, 2020-06-08:
I think I goofed when allowing the program through Windows controlled file access after uninstalling V1.3 and installing V1.2
Having allowed it properly, the program is now able to format and write the image. Thanks. However, the V1.3 filter may need a bit of tweaking?

maxnet, 2020-06-08:
Does v1.2 show the size of the SD card correctly in the SD card selection dialog?

In v1.3 we filter 3 more things compared to v1.2:

1) devices which report a zero size.
2) devices which are of a virtual drive type instead of a physical drive.
3) mac os x only: timemachine backup devices

If it is not being filtered because card reports zero size, it is probably that your card reader is incorrectly reporting as virtual drive.
If that turns out to be a problem in practice, we could remove that check in a future release.

(Reason the check for virtual devices was added is that someone on the forum mentioned that his software RAID volumes were showing up in Imager, and I thought it was better to filter those out.)

lurch, 2020-06-08:
> (Reason the check for virtual devices was added is that someone on the forum mentioned that his software RAID volumes were showing up in Imager, and I thought it was better to filter those out.)

Yeah, it seems there's so much variety in what various systems do and don't report (especially when you include the huge diversity of USB card-readers - some of which report themselves as single-slot removable devices, others which report themselves as multi-slot removable devices, and others which just appear as a "plain flash drive") that I suspect there's no "perfect solution" to this, just a "solution which is least annoying to the most people" :wink: :floppy_disk:

maxnet, 2020-06-09:
>especially when you include the huge diversity of USB card-readers

Well, at least the USB ones speak somewhat the same protocol.
The internal ones in laptops are sometimes hooked up through buses other than USB as well, and may be using custom drivers.

lurch, 2020-06-09:
In Linux-land, internal SD card-readers that are connected via USB tend to show up as `/dev/sdX` (where X is a lower-case letter) and internal card-readers that are connected via PCI tend to show up as `/dev/mmcblkN` (where N is a decimal number). Although you probably already knew that :slightly_smiling_face: 
In my experience, when an SD card appears as `/dev/sdX` then it just appears as a MSD, but when an SD card appears as `/dev/mmcblkN` you can also access the card's CID & CSD registers (and can access other low-level commands?).

maxnet, 2020-06-09:
And in Windows land you can look at the "bus" listed in diskpart's "detail disk" (commands listed above), to see if that may be the case here.

commit 39abeb23 (2020-07-03): Improve drive listing

- Show write protected SD cards greyed out.
  Show message if user select one of those. Closes #59
- Do show virtual drive types on Windows.
  Do keep them hidden on Mac OS X as that seems to
  cover .dmg file and timemachine stuff. Closes #79
closed 2020-07-03 by maxnet