raspberrypi/rpi-imager #80: update to rpi-imager 1.3-1 error...

hughparker1, 2020-06-09:
Manjaro Xfce error message...

-- Configuring incomplete, errors occurred!
See also "/var/tmp/pamac-build-hugh/rpi-imager/src/build/CMakeFiles/CMakeOutput.log".
==> ERROR: A failure occurred in build().
    Aborting...

maxnet, 2020-06-09:
pamac-build ?
Are you using 3rd party build scripts?

hughparker1, 2020-06-09:
[CMakeOutput.log.bak.txt](https://github.com/raspberrypi/rpi-imager/files/4751928/CMakeOutput.log.bak.txt)

> pamac-build ?
> Are you using 3rd party build scripts?

I updated using pamac gui

maxnet, 2020-06-09:
>I updated using pamac gui

Which is a front-end to build packages from source?
May want to fill a bug report to whoever is maintaining that...

Do you get a more clear error message if you run cmake yourself?
Among the lines of:

```
git clone --depth 1 https://github.com/raspberrypi/rpi-imager.git
cd rpi-imager
cmake .
```

(If you make any changes to rectify errors occured when running cmake, make sure you "rm CMakeCache.txt" before running "cmake ." again)

hughparker1, 2020-06-09:
> Which is a front-end to build packages from source?
> May want to fill a bug report to whoever is maintaining that...
>
> Do you get a more clear error message if you run cmake yourself?
> Among the lines [...]

thanks for reply, I will remove "CMakeCache.txt" then run the three commands you suggested.

I'm not very technical but I'll give it a try and let you know the outcome.

hughparker1, 2020-06-09:
I removed "CMakeCache.txt" then ran the three commands you suggested, but didn't work, I've never used "cmake" before.  Do I need to specify a path after cmake  ? ....
[cmake.txt](https://github.com/raspberrypi/rpi-imager/files/4752127/cmake.txt)

maxnet, 2020-06-09:
Use "cmake ." including dot.
The dot means the path to the "current directory".

hughparker1, 2020-06-09:
> Use "cmake ." including dot.
> The dot points to the path of the "current directory".

oops, my mistake, I'll try again

hughparker1, 2020-06-09:
I ran the command again but received an error (see output below). didn't any log file...
this is all new to me so not sure what's going on tbh
[cmake2.txt](https://github.com/raspberrypi/rpi-imager/files/4752171/cmake2.txt)

hughparker1, 2020-06-09:
sorry these is a log, but in a different place...
[CMakeOutput.log.bak2.txt](https://github.com/raspberrypi/rpi-imager/files/4752204/CMakeOutput.log.bak2.txt)

maxnet, 2020-06-09:
That has a better error message:

```
-- Could NOT find Qt5LinguistTools (missing: Qt5LinguistTools_DIR)
```

A difference between Imager 1.2 and 1.3 is that we now support internationalization.
If a user with a Chinese operating system starts Imager, he will be presented with a Chinese user-interface instead of English.

In order to enable that we have to run some extra commands as part of the build process (lrelease), and those are not installed on your system.
On Debian based systems the package that provides the necessary commands is called qttools5-dev.
But I don't know what the package is called on your operating system, and how to install that.
Better ask the person that maintains the rpi-imager pamac build script for advice...

hughparker1, 2020-06-09:
> That has a better error message:
>
> ```
> -- Could NOT find Qt5LinguistTools (missing: Qt5LinguistTools_DIR)
> ```
>
> A difference between Imager 1.2 and 1.3 is that we now support internationalization.
> If [...]

OK thanks for your advice

maxnet, 2020-06-09:
>EDIT2: I'm not sure who maintains the rpi-imager pamac build script, I thought this forum was the 
>best place to ask for help.

Problem is thinks like which packages you need to install to build this program from source code is specific to your Linux distribution used.
And I don't think there are many Manjaro users hanging out here.
(If I am wrong and somebody else subscribed to notifications in this repository is using it, feel free to speak up).

If Pamac/Manjaro is using ArchLinux packages as base (not certain if they do, it is more an educated guess), then you can find the e-mail address of the package maintainer here: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=rpi-imager
Recall you can also report bugs through that webinterface, but they made it ridiculously complicated to do so.
The anti-spam challenge on the account registration page requires that you run a command that is only available if you actually run Archlinux (or a derivative)
So I cannot do that.

hughparker1, 2020-06-09:
Many thanks for your help, much appreciated

lurch, 2020-06-09:
> I'm not sure who maintains the rpi-imager pamac build script

I've never used Manjaro myself, but a bit of searching reveals it's based on Arch Linux (which I've never used either).
Further searching found https://aur.archlinux.org/packages/rpi-imager/ - maybe you can file a bug-report or similar over there? :shrug:

maxnet, 2020-06-12:
Arch git log:

>Age | Commit message (Expand) | Author
>3 days | Add missing qt5-tools makedepends

@hughparker1 can you confirm issue is fixed now?

hughparker1, 2020-06-14:
Hi, sorry for late reply
I updated using pamac gui and system is now up-to-date!
Thanks for your help
Hugh

maxnet, 2020-06-14:
Good to hear it is solved.

closed 2020-06-14 by maxnet

sysdl132, 2020-10-06:
@maxnet 
Debian quick install (add repository) script:
```
curl -s https://packagecloud.io/install/repositories/sysdl132/rpi-imager/script.deb.sh | sudo bash
```