raspberrypi/rpi-imager #82: Imager goes above 100%

Bombardier-C-Kram, 2020-06-22:
[image]
Im not really sure what I did... I did cancel some OS installs a few times and now it goes a lot above 100%. Is this a known bug?

maxnet, 2020-06-22:
Linux platform?

Bombardier-C-Kram, 2020-06-22:
Yeah, Ubuntu 20.04

Bombardier-C-Kram, 2020-06-22:
I also noticed that if i tried to use etcher, I would get really slow speads... Like speads of just 90mb per second or something... Like 1 hour to do one flash of raspberry pi OS

maxnet, 2020-06-23:
Probably still writing out the previous write sessions you cancelled in the background, and that does not go well with the write progress code that asks kernel how much has been written to the SD card block device.

"Cancel" is not really effective on Linux, as it has huge write buffers.
Once Imager has copied everything that needs to be written to the kernel's buffers -which can occur within seconds if you are writing a local file- there is no effective way for us to cancel the operation.

Bombardier-C-Kram, 2020-06-23:
Ah okay. I just did it on a windows partition and it all worked well. Thanks

lurch, 2020-06-23:
> Once Imager has copied everything that needs to be written to the kernel's buffers -which can occur within seconds if you are writing a local file- there is no effective way for us to cancel the operation.

Would using `O_DIRECT` help with that? :shrug:

maxnet, 2020-06-23:
>Would using O_DIRECT help with that?

Yes, but then we wouldn't be able to take advantage of the kernel's highly efficient code to write from buffer to device.

Using O_DIRECT by-passes the kernel in talking to devices, and using it is pretending we know better how to pass data to devices than the kernel does.
Ask the average kernel developer what they think about that :-)

http://lwn.net/2002/0516/a/lt-deranged-monkey.php3

>The thing that has always disturbed me about O_DIRECT is that the whole
>interface is just stupid, and was probably designed by a deranged monkey
>on some serious mind-controlling substances

(Quote from Linus T.)

lurch, 2020-06-23:
I assumed for large-block-writes that it'd be beneficial to bypass the disk caches (see e.g. https://github.com/balena-io/etcher/issues/1523 ), but it sounds like you understand this much better than I do :slightly_smiling_face: 
(it was **ages** ago that I looked at all this, so perhaps I'm getting my flag names / nomenclature muddled up)

Bombardier-C-Kram, 2020-06-23:
Ok so its a linux problem?

maxnet, 2020-06-23:
>I assumed for large-block-writes that it'd be beneficial to bypass the disk caches

Maybe if one has code that is more optimized for that purpose than what we have now.
But right now Imager is much faster under Linux than under Windows and Mac OS X that have practically no write caching.

> I assumed for large-block-writes that it'd be beneficial to bypass the disk caches
> (see e.g. [balena-io/etcher#1523](https://github.com/balena-io/etcher/issues/1523) )

We do currently use O_DIRECT for *reads* (enable it through fcntl after the write) to prevent the read from cache problem.
Although it would probably be better to see if there is another way to drop cache (perhaps through posix_fadvise?) as well, as that is far from ideal, as it also disables read-ahead caching.
(Dropping block cache through /proc/sys/vm is not possible, as we are not running as root)

>Ok so its a linux problem?

That progressbar does not get over 100% is fixable. But accurate progress indication is a challenge under Linux, yes.

lurch, 2020-06-23:
Yeah, I figured you'd probably have a much better understanding of this than my vague recollections :laughing: 

> Probably still writing out the previous write sessions you cancelled in the background, and that does not go well with the write progress code that asks kernel how much has been written to the SD card block device.

Hmm, so just out of curiosity: does that mean if you start a write, cancel it and then start a second write, Linux is actually still writing out the first image onto the SD card shortly before overwriting it with the second image?

maxnet, 2020-06-23:
> Hmm, so just out of curiosity: does that mean if you start a write, cancel it and then start a second write, Linux is actually still writing out the first image onto the SD card shortly before overwriting it with the second image?

On my own Kubuntu 18.04 Linux box pressing "cancel" in Imager will result in the old image being written to disk first -to the extend it was already written to disk cache-, before it even returns to the user interface, and allows you to select a new image.
It seems to implicit flush everything to disk upon us closing the file descriptor with close().

If an implicit flush on close does not happen with the kernel version the TS is using, I would expect whether it writes the old image to SD card first to depend on how fast the new write occurs.
If the new write is about writing a local image file to SD card and the TS has a lot of memory, I would expect the entire block cache to be replaced very fast.
Once information in the block cache is replaced, Linux will no longer write the older information to disk.
However if things are not so fast, for example because it has to download the new image from Internet, it may indeed continue to write blocks from the old image in the background, yes.

lurch, 2020-06-23:
I wonder whether delaying the start-of-writing by a small amount, and using the delay to see if "something" is still being written to the SD card, would allow you to determine whether an old write-in-progress is still occurring? And thus you could "wait until the SD card settles" before starting the new write, which would enable you to calculate more accurate progress statistics?
Hmmm, but then again if a multi-gigabyte image is already in the write-cache, I guess you probably want to get the new image into the write-cache as soon as possible, such that it "invalidates" the old data.
Hmmm number 2: If you started flashing an 8GB image to an SD card (and assuming your Linux box has LOADS of free memory), and the whole 8GB got loaded into the write-cache, and then you cancelled the 8GB flash (with it continuing in the background), and you then selected a different 4GB image to flash, would Linux then load the 4GB image into the write-cache but end up writing the first 4GB of the second image onto the SD card, followed by the second half of the 8GB image? IYSWIM :wink:

maxnet, 2020-06-23:
>Hmmm number 2: If you started flashing an 8GB image to an SD card (and assuming your Linux box 
>has LOADS of free memory)

By default Linux will block on executing write(), and wait before continuing until everything is flushed, once dirty page buffers ("write cache") has grown beyond 20% of total memory (changeable through /proc/sys/vm/dirty_ratio )
So only if the system had 40 GB of memory, would there be a chance the whole 8 GB would be in memory.

Hmm, thinking about that. 
The 1282% while writing Raspberry Pi OS (which is quite some GB) the TS gets should then not be possible to be caused by this particular problem.
Might be something else.
I do see an internal SD card reader is used to write, perhaps that device is reporting statistics in a different way then Imager is expecting.
Not sure what exactly that could be though (different block size? perhaps the TRIM/DISCARD commands we sent result incorrect numbers? or something else, but hard to debug not having the same hardware).

Bombardier-C-Kram, 2020-06-23:
The SDCard reader is the one from the pi desktop kit... Also, It would
kinda hang and not even respond if I tried to kill it. As soon as I took it
out of the sd card slot, it killed.

maxnet, 2020-06-23:
>The SDCard reader is the one from the pi desktop kit... 

So a USB one?
Was a bit confused by "INTERNAL SD CARD" in your screenshot

Bombardier-C-Kram, 2020-06-23:
No, Like an sd card with a MicroSD in the back. The one in the official
desktop kit.

lurch, 2020-06-23:
> No, Like an sd card with a MicroSD in the back.

Ahhh, that's just an adaptor, used to convert the pins from a microSD card to a full-size SD card. The "reader" @maxnet is asking about is how is your SD-card-in-the-adaptor connected to your computer?

Bombardier-C-Kram, 2020-06-23:
Its connected in an sd card slot. I use a 2011 macbook pro which has an
internal sd card slot.

lurch, 2020-06-23:
> By default Linux will block on executing write(), and wait before continuing until everything is flushed, once dirty page buffers ("write cache") has grown beyond 20% of total memory (changeable through /proc/sys/vm/dirty_ratio )
So only if the system had 40 GB of memory, would there be a chance the whole 8 GB would be in memory.

Ahhh, thanks. Looks like I'm muddling up the way read buffers work (which **do** cache multi-GB files) with the way write buffers work (which are evidently much smaller!) :upside_down_face:

Bombardier-C-Kram, 2020-06-24:
So is there a fix to this problem or like not really?

lurch, 2020-06-24:
From what I understand of what @maxnet has said, the "fix" is to simply wait for a while after cancelling the first write, before starting the second write. Are you still seeing progress figures over 100% or has it settled back to normal again? If things are still "funny", have you tried doing a reboot?

Bombardier-C-Kram, 2020-06-24:
I’ve rebooted lots of times and waited quite a few hours... I saw it get to 100k percent so idk...I’m just using etcher rn but I just wanted to know if there is a fix for it. I have rebooted a lot of times so far.

maxnet, 2020-06-25:
>I’ve rebooted lots of times and waited quite a few hours...

Is the problem of it going beyond 100% present even if you do NOT "cancel" any writes, but only do a single write session after boot?

What output do you get if you execute "cat /sys/class/block/mmcblk0/stat" twice, once before you start the write, and a second time after the write is complete?

closed 2020-06-29 by maxnet

kwebber3, 2022-05-05:
I have a similar problem on Windows 11, I restarted the install at 124%. RPI imager v 1.7.2 installing a version from a zip downloaded here. https://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-14/

Dustie, 2023-05-11:
Just adding that I also see this happening in Windows 11, with a OPNsense image and a good USB 3 stick.

maxnet, 2023-05-11:
>Just adding that I also see this happening in Windows 11, with a OPNsense image and a good USB 3 stick.

Don't think we offer that in the repository.
Got a download link?

maxnet, 2023-05-11:
>Well, the image wasn't actually for the RPI. I just used the RPI Imager as I had it on hand. It was the nano OPNsense 
>image:
>
>https://opnsense.org/download/

Did you extract the .bz2 file first, before opening it in Imager?
(we do not compile in libbzip2 support on the Windows platform. So it should normally fail if you open the .bz2 directly. But libarchive may still be able to handle it if it notices you have a bzip2.exe in your path, and can outsource uncompressing to that. Have not tested that though).

Cannot reproduce the issue with Windows 11 under Virtualbox with opnsense-23.1-openssl-nanon-amd64.img (extracted):

[image]

Dustie, 2023-05-11:
I did extract it first, yes. I can't reproduce it myself either. If I can add helpful input I will of course, but I'm not here expecting a fix or anything. Tool works great. I just found it odd with 200%, googled it, and ended up here :)

Lopastudio, 2024-03-10:
same issue here
I used HamPi rpi image, and I am running windows 11 22h2.
[image]

commit 83ae3f1c (2020-06-29): Linux: do not trust device write stats if higher than what we wrote

For reasons unknown the device write statistics sometimes report
higher numbers than the number of bytes we wrote to the device on
some hardware. Causing write percentage in progressbar text
to exceed 100%

Maximize amount of bytes to what we wrote to the device.
This do will result in unreliable write progress on these devices
(it will then show amount of bytes written to cache instead of
what made it to device yet)

Closes #82