raspberrypi/rpi-imager #83: Add openEuler-raspi image link

woqidaideshi, 2020-06-24:
I’m the maintainer of openEuler RaspberryPi group. Our group has migrated openEuler to Raspberry Pi 3B/3B+/4B and built an openEuler-RaspberryPi image. openEuler is an open-source and optimized Linux distribution platform and is popular in China (https://openeuler.org/en/). Shall we add the openEuler-RaspberryPi image into Raspberry PI's third-party image download page (i.e., https://www.raspberrypi.org/downloads/), just as that of “Raspberry Pi Imager for Ubuntu" or "Ubuntu MATE"? Thanks.

woqidaideshi, 2020-07-08:
Main repository of openEuler RaspberryPi: https://gitee.com/openeuler/raspberrypi/blob/master/README.en.md. You can find openEuler-RaspberryPi image download link there.

The 4.19.90 openEuler kernel for running on Raspberry Pi: https://gitee.com/openeuler/raspberrypi-kernel/blob/master/README.en.md.

ghollingworth, 2020-07-08:
We would rather add it to the rpi-imager application, to do this we would need you to provide a .json file containing the options for installation and logos etc. to suitably fit into the rpi-imager menu.

http://downloads.raspberrypi.org/os_list_imagingutility.json

This is our top level .json file which describes the main menu.  You would need to provide a section for the openEuler image, plus provide your .json file equivalent to Ubuntu's here:

https://changelogs.ubuntu.com/raspi/os_list_imagingutility_ubuntu.json

Do you have any download stats which show the use of the OS in China?

Thanks

woqidaideshi, 2020-07-17:
Thanks for your kind reply. We would provide this two .json files and download stats after our new stable version is released in September this year. 
Thanks.

woqidaideshi, 2020-11-10:
@ghollingworth 

Our section for the openEuler image in the top level .json file is as follows:

{
    "os_list": [
        {
            "name": "openEuler",
            "description": "A unified and open OS that supports multiple processor architectures",
            "icon": "https://openeuler.org/Raspberry_Pi_OS_openEuler.png",
            "subitems_url": "https://openeuler.org/os_list_imagingutility_openeuler.json"
        }
    ]
}

Our .json file equivalent to Ubuntu's here:
<https://openeuler.org/os_list_imagingutility_openeuler.json>.
```
Besides, we have a brief introduction of openEuler to show the use of the OS in China:
openEuler is an open source, free Linux distribution platform. The platform provides an open community for global developers to build an open, diversified, and architecture-inclusive software ecosystem. 
In March 2020, the openEuler community has released its first Long-Term Support (LTS) edition, the openEuler 20.03 LTS. In the past 7 months, the ISO had seen more than 16,000 downloads. There are currently 73 Special Interest Groups (SIGs) and more than 1,200 contributors in the openEuler community.
```

woqidaideshi, 2020-11-20:
@ghollingworth
Could you help me review my last comment above and add our section to the top level .json file?

By the way, I have two questions as follows:
1. When and where can I see my submitted information on the official website of Raspberry Pi? 
2. Can users download the openEuler image directly from the rpi-imager application after our section for the openEuler image is added to the top level .json file?

Thanks.

ghollingworth, 2020-12-02:
Hi, sorry I've not replied, we're in the process of fixing up rpi-imager to add some extra functionality so these alternative operating systems can be given a go, we're also going to monitor downloads so we can see how each OS is being downloaded.

Will be able to add it as soon as rpi-image v1.5 is released (which should be in the next couple of days)

woqidaideshi, 2020-12-03:
Thanks a lot for your reply. Besides downloading from the rpi-image application, can users download our openEuler image from the official website of Raspberry Pi? I mean, is there anywhere else on the official website of Raspberry Pi where users can download the openEuler image directly?

Thanks.

ghollingworth, 2020-12-15:
I've now updated the Raspbery Pi Imager to v1.5 which includes support for
sub-menus.  Which means I can bring the OpenEuler release in.  Is this a
Chinese specific distribution?  It looks like it is and should probably be
mentioned in the spiel for the OS?

Thanks

Gordon

woqidaideshi, 2020-12-17:
@ghollingworth
Thanks for your help.

The openEuler release is not a Chinese specific distribution. It is an English distribution that aims to be used by the people all over the world.

Maybe the current openEuler description is too simple, we can change it to `A unified and open OS that targets servers with multiple processor architectures.` In other words, our section for the openEuler image in the top level .json file will be as follows:

```
{
    "os_list": [
        {
            "name": "openEuler",
            "description": "A unified and open OS that targets servers with multiple processor architectures.",
            "icon": "https://openeuler.org/Raspberry_Pi_OS_openEuler.png",
            "subitems_url": "https://openeuler.org/os_list_imagingutility_openeuler.json"
        }
    ]
}

Accordingly, we modify the first sentence of the openEuler brief introduction and the new version of the brief introduction is as follows:

```
openEuler is an open source, free Linux distribution platform, especially for servers. The platform provides an open community for global developers to build an open, diversified, and architecture-inclusive software ecosystem. In March 2020, the openEuler community has released its first Long-Term Support (LTS) edition, the openEuler 20.03 LTS. In the past 7 months, the ISO had seen more than 16,000 downloads. There are currently 73 Special Interest Groups (SIGs) and more than 1,200 contributors in the openEuler community.

P.S. the first character `o` of openEuler is a lower case.

Thanks.

ghollingworth, 2020-12-17:
OK, I had assumed it was a special Chinese specific build that would help Chinese users get into Raspberry Pi.  Otherwise, I don't see how it stands out against the other tens of Linux distributions for Raspberry Pi.

woqidaideshi, 2020-12-18:
Yes, openEuler community will help more Chinese users to use Raspberry Pi. At present, many universities and colleges in China are using the openEuler + Raspberry Pi as the experimental environment of their OS courses.

Currently, many Chinese developers are contributing code to openEuler. Compared to other Linux distributions, openEuler focues on and has performed a lot of optimization on AArch64 and RISC-V, as well as x86. For example, some key features/optimizations of openEuler 20.03 LTS are listed in <https://openeuler.org/en/docs/20.03_LTS/docs/Releasenotes/key-features.html>. In this year, some SIGs in openEuler community like cloud-ai, are collaborating with other upstream communities to improve the upstream software running on AArch64 arch.

woqidaideshi, 2020-12-18:
@ghollingworth 
Yes, openEuler community will help more Chinese users to use Raspberry Pi. At present, many universities and colleges in China are using the openEuler + Raspberry Pi as the experimental environment of their OS courses.

Currently, many Chinese developers are contributing code to openEuler. Compared to other Linux distributions, openEuler focues on and has performed a lot of optimization on AArch64 and RISC-V, as well as x86. For example, some key features/optimizations of openEuler 20.03 LTS are listed in <https://openeuler.org/en/docs/20.03_LTS/docs/Releasenotes/key-features.html>. In this year, some SIGs in openEuler community like cloud-ai, are collaborating with other upstream communities to improve the upstream software running on AArch64 arch.

woqidaideshi, 2020-12-21:
@ghollingworth
I'm sorry that I downloaded the latest version (the Raspbery Pi Imager v1.5) but cannot find our openEuler image in it. Could you help me figure out when our openEuler image will be added into the rpi-image? Thanks!

ghollingworth, 2020-12-21:
It has not yet been added, I spent a little more time researching it and didn't realise it's the commercial version of EulerOS.  As such we need to get it checked by our lawyers first.

woqidaideshi, 2020-12-22:
@ghollingworth

The openEuler OS is not commercial, but open and free. EulerOS is a commercial distribution of openEuler. Their relationship is similar with that of openSUSE and SUSE Linux Enterprise.

Most software in openEuler use the same open-source licenses as their upstream projects, such as [openEuler kernel](https://gitee.com/openeuler/kernel). Others use [Mulan license](https://gitee.com/openeuler/raspberrypi/blob/master/LICENSE) which has been approved by OSI, such as [the openEuler RaspberryPi project](https://gitee.com/openeuler/raspberrypi). Mulan license allows developer to use and modify the code for both private and commercial use. 

P.S., the source code of all software in openEuler is published and developed on <https://gitee.com/openeuler/> and <https://gitee.com/src-openeuler/>.

tdewey-rpi, 2023-11-17:
Please fill out a request to be added to the OS list using this form: https://forms.gle/pr32R8vVYadRuGha7

Thanks,

Tom

closed 2023-11-17 by tdewey-rpi