raspberrypi/rpi-imager #87: Windows10 : Freeze when trying to select SD Card.

Simsimpicpic, 2020-07-02:
The RPI-IMAGER doesn't respond when trying to select SD Card...

[image]

```xml
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Application Hang" /> 
  <EventID Qualifiers="0">1002</EventID> 
  <Level>2</Level> 
  <Task>101</Task> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2020-07-02T14:47:41.758887100Z" /> 
  <EventRecordID>13632</EventRecordID> 
  <Channel>Application</Channel> 
  <Computer>LAPTOP-7C0G0A32</Computer> 
  <Security /> 
  </System>
<EventData>
  <Data>rpi-imager.exe</Data> 
  <Data>1.3.0.0</Data> 
  <Data>40f4</Data> 
  <Data>01d6507f5a6653fc</Data> 
  <Data>60000</Data> 
  <Data>C:\Program Files (x86)\Raspberry Pi Imager\rpi-imager.exe</Data> 
  <Data>2e78d7b9-68eb-48d1-88f9-99994251dce5</Data> 
  <Data /> 
  <Data /> 
  <Data>Unknown</Data> 
  <Binary>55006E006B006E006F0077006E0000000000</Binary> 
  </EventData>
</Event>
```

Thank you.

maxnet, 2020-07-02:
Can you properly use the SD card in explorer? E.g. put files on SD card.
Seen that before with broken SD cards (or broken SD card holders).

Under Windows the code that enumerates the available drives and list their sizes seems to actual trigger real-time communication with the SD card (unlike other operating systems where the kernel gets the data once when the card is inserted, and returns cached information to applications).
If there is a problem with the SD card the functions hangs for a long time until a timeout elapses.
Since we are currently calling the function in the UI thread, that also causes the UI to hang.
Could move the code the enumeration code to different thread to prevent that.
However that will still mean the SD card is not good to use.

Simsimpicpic, 2020-09-27:
With the new update, i no longer have this problem.
Thank you 👍🏻

closed 2020-09-27 by Simsimpicpic