raspberrypi/rpi-imager #88: Application can't be tracked by MacUpdater

ulfklose, 2020-07-05:
I got in contact with the creators of MacUpdater and asked them why the rpi-imager isn't being tracked by their application.

The possible reason is explained here: https://www.corecode.io/macupdater/faq.html#SHOWUP

I don't know of this is of any relevance for you guys but I'd like to ask you, if possible, to fix this.

maxnet, 2020-07-05:
>The possible reason is explained here: https://www.corecode.io/macupdater/faq.html#SHOWUP

"If you want to find out why a particular app does not show up in MacUpdater, you can drag the app onto MacUpdater's main window to get an explanation."
So what does that give for you?

ulfklose, 2020-07-05:
Sorry, my bad, I could have done this right on creating this ticket.

[image]

core-code, 2020-07-05:
this means both the CFBundleVersion and CFBundleShortVersionString entries in the Info.plist are missing.

this is not just a problem with regards to MacUpdater but:

• not having a version number is a violation of Apple's macOS development guidelines
• if you have multiple versions installed, macOS will be unable to determine which version is the newest one
• also if you select the app in the Finder (via Get Info, Column View or Quick Look) you will not be able to inspet the version number (because it is missing). Mac users are accustomed to be able to inspect the version numbers of any app in the Finder.

maxnet, 2020-07-05:
Thanks for the info.
Will populate those in the .plist.

closed 2020-07-05 by maxnet

core-code, 2020-07-17:
thanks i can confirm that the 1.4 release here:
https://github.com/raspberrypi/rpi-imager/releases/download/v1.4/imager.dmg

fixes the issue. however the download from the homepage ( https://www.raspberrypi.org/downloads/ ) doesn't seem to be updated yet.

maxnet, 2020-07-18:
>thanks i can confirm that the 1.4 release here:
>fixes the issue

Good to hear.

>however the download from the homepage ( https://www.raspberrypi.org/downloads/ ) doesn't seem to be updated yet.

Has been updated now.
Can be a slight delay due to different people in charge of the github page (where releases are basically only created so that the code in the repository is tagged correctly), downloads.raspberrypi.org and www.raspberrypi.org

commit f90739a5 (2020-07-05): Mac: populate version info in .plist

Closes #88