raspberrypi/rpi-imager #92: SD card not recognised on Chromebook

TMorgan99, 2020-07-25:
Not sure why, but the Imager cannot see the SD card on my Chromebook.  Any help with this? Thanks!

maxnet, 2020-07-25:
Chromebook?
Are you sure you are using the Raspberry Pi Imager software, and not the Chromebook recovery utility program?

TMorgan99, 2020-07-26:
Yes, I downloaded the imager.
I can access the SD from the Files, where I was able to format it, but
inside imager, there are no options for SD card.
Here's the screenshot.

maxnet, 2020-07-26:
Does running "lsblk --bytes --json --paths --output-all" show the card?

TMorgan99, 2020-07-26:
Thanks for your suggestion.

The following is produced. With the card inserted or not, no difference, so
it looks like it's not visible to lsblk.

tmorgan99@penguin:~$ lsblk --paths
NAME       MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
/dev/vda   254:0    0 51.8M  1 disk /opt/google/cros-containers
/dev/vdb   254:16   0 16.1G  0 disk /dev/bus/usb/002/001
/dev/pmem0 259:0    0  176M  0 disk /dev/.ssh/sshd_config

But it does show up in the /mnt tree.
# without
tmorgan99@penguin:/mnt/chromeos$ ls
fonts
# with
tmorgan99@penguin:/mnt/chromeos$ ls
fonts  removable

lurch, 2020-07-27:
> Here's the screenshot.

FYI attachments don't work on Github when replying via email, you can only do that via the github website.

> so it looks like it's not visible to lsblk.

> But it does show up in the /mnt tree.

Weird - the Chromebook is probably doing something "special", which standard tools don't recognise. What does the `mount` command say?

TMorgan99, 2020-07-27:
yes, it is weird.
There is no difference to mount, 66 lines each way.
[image]
[image]

And here, I attach the screenshots via Github's markdown so you dont miss them.
Thanks again!

maxnet, 2020-07-27:
Not supported by Chrome OS: https://bugs.chromium.org/p/chromium/issues/detail?id=980456

TMorgan99, 2020-07-27:
Great, thanks. I was thinking it was further upstream, now I know where to
go.

KInda disappointing, since the Chromebook has an SD card as a sales
feature,,,

rayfoss, 2020-10-17:
To sum it up: 

- watch issue https://bugs.chromium.org/p/chromium/issues/detail?id=980456
- USB passthrough is only supported for android phones USB Debugging...
- Android Apps can't get block level access
- There is NO way to flash an image to a block device on ChromeOS... except using the NaCL Chromebook Recovery tool... or jumping though senseless hoops like dual booting etc

The only utility that works is here:
https://chrome.google.com/webstore/detail/chromebook-recovery-utili/jndclpdbaamdhonoechobihbbiimdgai

Why is it taking so long? it can be a real security hazard, not just to you, but to others... imagine borrowing a usb keyboard and having a freak rogue app flash custom spy firmware onto it. Or having your built in hardware that runs over USB get compromised... such as the sd card readers... webcam... etc.

closed 2021-01-17 by maxnet