raspberrypi/rpi-imager #93: Impossible to write OS on sd card - Ubuntu 20.04

londer1995, 2020-07-26:
Hello, I got a couple of problems using rpi-imager that I never noticed before (maybe because it worked fine until now!).
First in first, I got this in the terminal:
[image]
It seems ok, but when it check the card it shows:
[image]
I can't be sure 100% (actually I'm not able to verify it!!) card is not broken, but it worked until 1 hour ago without problems.
Since I've tryied to change OS from raspberry OS to Ubuntu, I started having issues.
Is it related to rpi-imager? How can I verify the integrity of the SD card/rpi-imager? thx

maxnet, 2020-07-26:
Can you read the whole card using other tools?
E.g.

```
sudo dd if=/dev/sdX of=/dev/null bs=512K
```

Where sdX is your SD card reader.

londer1995, 2020-07-26:
[image]

I went to buy another SD card (samsung evo plus 64gb) and got the same error.
Translated error message is:
dd: error while reading '/dev/sdb' : input/output Error.

maxnet, 2020-07-26:
> [image]
> 
> I went to buy another SD card (samsung evo plus 64gb) and got the same error.
> Translated error message is:
> dd: error while reading '/dev/sdb' : input/output Error.

Well, if other tools like dd also give IO errors while reading from the card, than the problem is not in our software.

As far as another card having the same problem.
Problem does not necessarily have to be the SD card itself. Can also be something wrong with the reader, or the full-size SD card<->Micro SD adapter if you use any.

> EDIT: using sudo rpi-imager I get:
> `QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root' `

Warnings like that can be safely ignored.

londer1995, 2020-07-26:
Well, I'm using the new adapter with the new SD card... I'm going to format and install again Ubuntu, last chance before throwing everything out the window!!

Btw thank you, I'll edit this answer in about 30 mins just for completeness

closed 2021-01-17 by maxnet