raspberrypi/rpi-imager #94: Imager Crashes with BSOD Windows 10 v2004

AubsUK, 2020-07-27:
Fully up to date Windows 10 install, new download of Raspberry Pi Imager V1.4.0.0

Tried on a second laptop, same issue.  Both Dell.

Rebooted, ran as Administrator.
All BSODs appear to point to memory corruption, different errors every time, most common has been "PAGE FAULT IN NONPAGED AREA".

Issue appears to be at the end of validating:

- One one attempt with a 64GB SanDisk Ultra, writing gets to 100%, vaidating gets to 100%, then goes to 99% writing, then crashes.
- Another attempt with a 32GB SanDisk Ultra, writing and validating gets to 100%, on validation complete, click the button then BSOD again.

Not sure if the following crash dump analysis is of use to you, from the second attempt above:

```
Microsoft (R) Windows Debugger Version 10.0.19041.1 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Windows\MEMORY.DMP]
Kernel Bitmap Dump File: Full address space is available

Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 19041 MP (8 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Built by: 19041.1.amd64fre.vb_release.191206-1406
Machine Name:
Kernel base = 0xfffff807`82a00000 PsLoadedModuleList = 0xfffff807`8362a310
Debug session time: Mon Jul 27 21:00:59.742 2020 (UTC + 1:00)
System Uptime: 0 days 0:10:26.930
Loading Kernel Symbols
...............................................................
................................................................
................................................................
.............
Loading User Symbols

Loading unloaded module list
..............
For analysis of this file, run !analyze -v
0: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

KERNEL_MODE_HEAP_CORRUPTION (13a)
The kernel mode heap manager has detected corruption in a heap.
Arguments:
Arg1: 0000000000000011, Type of corruption detected
Arg2: ffff840a12202100, Address of the heap that reported the corruption
Arg3: ffff840a1fddcb50, Address at which the corruption was detected
Arg4: 0000000000000000

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.Sec
    Value: 10

    Key  : Analysis.DebugAnalysisProvider.CPP
    Value: Create: 8007007e on NEON

    Key  : Analysis.DebugData
    Value: CreateObject

    Key  : Analysis.DebugModel
    Value: CreateObject

    Key  : Analysis.Elapsed.Sec
    Value: 10

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 70

    Key  : Analysis.System
    Value: CreateObject

BUGCHECK_CODE:  13a

BUGCHECK_P1: 11

BUGCHECK_P2: ffff840a12202100

BUGCHECK_P3: ffff840a1fddcb50

BUGCHECK_P4: 0

POOL_ADDRESS:  ffff840a1fddcb50 Nonpaged pool

FREED_POOL_TAG:  Mdl_

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

PROCESS_NAME:  System

STACK_TEXT:  
ffffc60c`573c1d78 fffff807`82f9bfa8 : 00000000`0000013a 00000000`00000011 ffff840a`12202100 ffff840a`1fddcb50 : nt!KeBugCheckEx
ffffc60c`573c1d80 fffff807`82f9c008 : 00000000`00000011 00000000`00000000 ffff840a`12202100 00000000`00000000 : nt!RtlpHeapHandleError+0x40
ffffc60c`573c1dc0 fffff807`82f9bc35 : 00000000`000000d0 ffff840a`1fda3000 00000000`00001000 00000000`00000000 : nt!RtlpHpHeapHandleError+0x58
ffffc60c`573c1df0 fffff807`82e27c54 : 00000000`00000001 00000000`00000001 00000000`00000000 00000000`00000008 : nt!RtlpLogHeapFailure+0x45
ffffc60c`573c1e20 fffff807`82c40a92 : ffff840a`12202340 00000000`000000ff 00000000`00000000 00000000`00000000 : nt!RtlpHpLfhSubsegmentFreeBlock+0x1c1034
ffffc60c`573c1ed0 fffff807`833c2019 : ffff840a`00000000 ffff840a`1fddcb60 fffffb64`80a9a008 01000000`00100000 : nt!ExFreeHeapPool+0x362
ffffc60c`573c1fb0 fffff807`9c028051 : ffffc901`53401000 ffffdc0f`195f61f0 ffff840a`1e1ad000 00000000`00000000 : nt!ExFreePool+0x9
ffffc60c`573c1fe0 fffff807`9bff4e33 : ffffc901`53401000 ffff840a`1e1ad000 ffffdc0f`158955c0 ffffdc0f`180e2be0 : dxgmms2!VIDMM_PAGE_TABLE_BASE::FreeCpuVisibleAddress+0x31511
ffffc60c`573c2010 fffff807`9bff03ce : ffff840a`1fddcb60 00001f80`01018bf0 00000000`00000002 00000000`00000200 : dxgmms2!VIDMM_PAGE_TABLE_BASE::FreeDriverUpdateAddress+0x37
ffffc60c`573c2040 fffff807`9c0282d2 : 00000000`00000000 ffffc901`53401000 00000000`00000002 ffffc60c`573c2560 : dxgmms2!VIDMM_PAGE_TABLE::UpdatePageTableInvalidate+0x10e
ffffc60c`573c2110 fffff807`9bff573d : 00000000`00000002 00000000`02800000 00000000`00000000 00000000`02800000 : dxgmms2!VIDMM_PAGE_TABLE::CommitVirtualAddressRange+0x316fa
ffffc60c`573c21b0 fffff807`9c0267d2 : 00000000`00000002 ffffdc0f`158955c0 ffffc901`4fd40101 00000000`00000000 : dxgmms2!VIDMM_PAGE_DIRECTORY::CommitVirtualAddressRange+0x77d
ffffc60c`573c2320 fffff807`9bff49ee : 00000000`00000000 ffffdc0f`158955c0 00000000`00001000 00000000`00000000 : dxgmms2!VIDMM_PAGE_DIRECTORY::CommitVirtualAddressRange+0x31812
ffffc60c`573c2490 fffff807`9c03dea2 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : dxgmms2!CVirtualAddressAllocator::CommitVirtualAddressRange+0x2f6
ffffc60c`573c2610 fffff807`9c025522 : 00000000`00000000 00000000`00000000 00000000`00000076 ffffdc0f`3f90b680 : dxgmms2!VIDMM_GLOBAL::InitPagingProcessVaSpace+0x1ce4e
ffffc60c`573c2780 fffff807`9c0182e2 : ffffdc0f`15895250 ffffc60c`00000001 00000000`00000001 ffffdc0f`194fbe50 : dxgmms2!VIDMM_GLOBAL::ProcessSystemCommand+0x31e22
ffffc60c`573c2900 fffff807`9c021b99 : ffffdc0f`194fbde0 ffff840a`1c72d001 ffffc60c`57271700 00000000`05f2da00 : dxgmms2!VIDMM_WORKER_THREAD::Run+0x1462
ffffc60c`573c2ae0 fffff807`82d46735 : ffff840a`1c72d080 fffff807`9c021b90 ffffdc0f`194fbde0 000fa4ef`bd9bbfff : dxgmms2!VidMmWorkerThreadProc+0x9
ffffc60c`573c2b10 fffff807`82de51b8 : ffffc901`4f951180 ffff840a`1c72d080 fffff807`82d466e0 00650064`00770073 : nt!PspSystemThreadStartup+0x55
ffffc60c`573c2b60 00000000`00000000 : ffffc60c`573c3000 ffffc60c`573bc000 00000000`00000000 00000000`00000000 : nt!KiStartSystemThread+0x28

SYMBOL_NAME:  dxgmms2!VIDMM_PAGE_TABLE_BASE::FreeCpuVisibleAddress+31511

MODULE_NAME: dxgmms2

IMAGE_NAME:  dxgmms2.sys

IMAGE_VERSION:  10.0.19041.388

STACK_COMMAND:  .thread ; .cxr ; kb

BUCKET_ID_FUNC_OFFSET:  31511

FAILURE_BUCKET_ID:  0x13a_11_Mdl__dxgmms2!VIDMM_PAGE_TABLE_BASE::FreeCpuVisibleAddress

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {73dcc25f-346e-904e-131c-c3ba37519271}

```

maxnet, 2020-07-27:
> Tried on a second laptop, same issue. Both Dell.

Do both have similar hardware?
E.g. same GPU?

> All BSODs appear to point to memory corruption, different errors every time, most common has been "PAGE FAULT IN NONPAGED AREA".

An user space program like Imager should not be able to corrupt kernel memory.

If you are an advanced user/adventurous, you may want to play with Driver Verifier to see if you can detect which driver is corrupting things: https://answers.microsoft.com/en-us/windows/forum/windows_10-update/driver-verifier-tracking-down-a-mis-behaving/f5cb4faf-556b-4b6d-95b3-c48669e4c983

AubsUK, 2020-07-27:
Thanks for the quick reply.  Below are the specs for the two machines.  Frustratingly, I followed all the config on the Driver Verifier.  I tried it twice and it was good both times, then I disabled it, rebooted and tried again, and it BSOD again.

I did notice that it took a lot longer to complete, assumed because it is verifying everything as it goes along.

I wonder if the app is doing something after verifying but whatever it is doing something (the disk) isn't ready at that point?

I noticed when it was running slowly:
- The disk is completely unallocated
- It gets to 100% writing
- It gets to Verifying 100%
- It goes to writing 99%
- the disk gets two volumes and the remaining unallocated space
- The disk is unmounted and the Write Successful message appears almost instantaneously. 

But when it ran at normal speed (pretty quick), I get the stop error before the disk gets any visible volumes.  After rebooting and looking at Disk Manager, it shows the two volumes and remaining unallocated space.

### Generated by Piriform Speccy v1.32.740 27 Jul 2020 @ 21:55
**Operating System**
Windows 10 Pro 64-bit
**CPU**
Intel Core i7 7700HQ @ 2.80GHz: 46 °C
Kaby Lake 14nm Technology
**RAM**
16.0GB Single-Channel Unknown (17-17-17-39)
**Motherboard**
Dell Inc. 065C71 (U3E1)
Graphics
Generic PnP Monitor (1920x1080@60Hz)
Intel HD Graphics 630 (Dell)
4095MB NVIDIA GeForce GTX 1050 Ti (Dell): 44 °C
ForceWare version: 445.75
SLI Disabled
**Storage**
232GB Samsung SSD 970 EVO 250GB (Unknown (SSD))
931GB Seagate ST1000LM035-1RK172 (SATA ): 35 °C
59GB SDXC Card (USB )
**Optical Drives**
No optical disk drives detected
**Audio**
Realtek Audio

### Generated by Piriform Speccy v1.32.740 27 Jul 2020 @ 21:53
**Operating System**
Windows 10 Pro 64-bit
**CPU**
Intel Core i7 @ 1.80GHz: 51 °C
Kaby Lake-U/Y 14nm Technology
**RAM**
16.0GB
**Motherboard**
Dell Inc. 0Y1T82 (U3E1)
**Graphics**
Generic PnP Monitor (1920x1080@60Hz)
Intel UHD Graphics 620 (Dell)
4096MB ATI Radeon 530 (Dell): 48 °C
CrossFire Disabled
**Storage**
1863GB SanDisk X400 M.2 2280 256GB (SATA ): 24 °C
238GB SK hynix SC311 SATA 256GB (SATA (SSD)): 30 °C
**Optical Drives**
No optical disk drives detected
**Audio**
Realtek Audio

lurch, 2020-07-27:
What anti-virus / security software do you have installed? If you temporarily disable it, does that stop RPi Imager from crashing?

maxnet, 2020-07-27:
Does the previous version (v1.3 from "releases" page on github) have same problem?

> I wonder if the app is doing something after verifying but whatever it is doing something (the disk) isn't ready at that point?

After verifying what has been written so far, it will seek back to the start of disk and write the first MB of the image that contains the partition table.
We skipped that at first, and saved that for last as the moment there is a partition table other Windows services (windows search indexer thing, virus scanners, etc.) may start accessing storage and cause all kinds of interesting problems if we were still in the process of writing the image.
Afterwards it will flush buffers, and eject the device.

AubsUK, 2020-07-30:
Thanks for all your help with this.  I have resolved the issue, sorry for the additional work.

Both devices have Realtek card readers, I will put it down to this being the cause.  I dug out an old USB card reader and it works fine (tested twice on first machine).

First laptop (assume the card was in when Speccy was run)
```
USB Root Hub (USB 3.0)
Realtek USB 2.0 Card Reader
SDXC Card
```

Second laptop (assume card was not in when Speccy was run)
```
USB Root Hub (USB 3.0)
Realtek USB 2.0 Card Reader
```

For full completeness in answering the questions:

Good thinking, I always forget to check this.  Microsoft Defender.  It was a PITA to disable, it would automatically re-enable itself.  Had to disable via group policy.  Same thing occurred.
> What anti-virus / security software do you have installed? If you temporarily disable it, does that stop RPi Imager from crashing?

Yes, same thing.
> Does the previous version (v1.3 from "releases" page on github) have same problem?

closed 2020-07-30 by AubsUK

lurch, 2020-07-30:
Might be worth checking your laptop manufacturer's support website to see if they have updated drivers for your built-in card readers?

AubsUK, 2020-07-30:
Both laptops are already fully updated for all drivers, frustratingly.  The Dell Support Assistant does a good job at this normally.  But have double checked and yes, it's the same.  No biggie really as an eternal card reader does the job.  And I've spent enough time on the built-in card reader to give up for now, but one day I may revisit it.

> Might be worth checking your laptop manufacturer's support website to see if they have updated drivers for your built-in card readers?

shinebayar-g, 2020-12-19:
I have a Dell laptop and getting similar BSOD near finishing, @AubsUK did you end up using external card reader?

AubsUK, 2020-12-19:
> I have a Dell laptop and getting similar BSOD near finishing, @AubsUK did you end up using external card reader?

I did, I used a USB card reader, all working perfectly.

shinebayar-g, 2020-12-19:
I see. I just made bootable Linux Live USB and successfully installed OS for my SD card. :)
f* windows :)

bedrin, 2020-12-24:
Same for me also on Dell laptop. Started just recently

elael, 2021-01-15:
I'm having the same issue with a Dell laptop too.

akbar-amin, 2021-01-29:
Same here. The USB reader method ended up working for me as well.

theonemcdonald, 2021-02-01:
I'm using an Anker 2in1 reader in addition to the built in reader in my Dell. Weird

theonemcdonald, 2021-02-01:
Well wouldn't you know it, the Anker uses the exact same chipset as the built in reader... Realtek garbage. Because these readers only present themselves to the system when a card is inserted, and if you’ve never used a Realtek reader on your computer, Windows Update or Dell Command can’t see the hardware and thus won’t bother updating the drivers.

TL;DR upgrade your drivers and try again.

maxnet, 2021-02-01:
>I'm using an Anker 2in1 reader in addition to the built in reader in my Dell.

Do you have an exact model number of that Anker reader?

(Did buy a cheap reader with RTL chipset the other day to see if I can reproduce the problem.
But received one with a newer chip, that does not need special drivers and does not have this particular problem).

theonemcdonald, 2021-02-01:
Model is Anker AR200 and it actually uses the same driver as the built in reader in my latitude 3510.

Seems like a race condition for those of us who have never used a card reader with this chipset on a new machine and the drivers haven’t been updated yet.

After letting windows update do its thing, I have no issues with RPi imager, Etcher, win32diskimager, etc...all of which would BSOD on both the internal and external reader. Both are working fine now

maxnet, 2021-02-03:
>Model is Anker AR200 and it actually uses the same driver as the built in reader in my 
>latitude 3510.

Unable to reproduce with an AR200.

Can confirm it indeed has special drivers.
But works fine for me with both the 2018 drivers that come with Windows 10, and the 2019 ones from Windows update.

[image]
[image]

That do is with a standard Chinese Teclast laptop, not with Dell.

So either Dell ships with another driver version that do has a problem, or the issue is dependent on timing or other factors.

AubsUK, 2021-06-26:
For completeness, I installed the latest version from Dell, but this still caused an issue, as did other versions from elsewhere, including inserting the card and running Windows Updates again.

Finally, I installed the "Card reader driver for XP,Vista,Win7,Win8,Win8.1,Win10" Version 10.0.370.188, dated 2018/03/13, from the following page and it now works without any issue.  Thanks for all the comments, hopefully this will help someone esle too.

https://www.realtek.com/en/component/zoo/category/card-reader-solutions-card-reader-controllers-software

RolfBly, 2021-11-25:
> For completeness, I installed the latest version from Dell, but this still caused an issue, as did other versions from elsewhere, including inserting the card and running Windows Updates again. [...]

Confirm on Dell Vostro. Also, very good of you to put a link to the correct answer up top. Saves quite a bit of time reading. Cheers!

dfpr, 2022-09-20:
Realtek site removed the file, I used this one to fix the issue for my Dell Inspiron 14 7472, neither windows update nor dell update offered this version dated 2021
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=6165w

Z3bc2G, 2025-02-27:
OMG you saved me !
Looks like my BSOD are gone <3  <3  <3
Thanks.

TCWORLD, 2025-03-24:
The above Dell driver link above also solves the BSOD problem for Dell Precision 5820 Tower devices (even though Dell says it's not supported).