raspberrypi/rpi-imager #95: Add ALT image link

gochankot, 2020-07-31:
I will represent the ALT Linux Team (https://github.com/altlinux). Our group maintains a port on aarch64 (https://en.altlinux.org/Ports/aarch64) in main architecture mode. Below, the presented ALT distributions were tested on Raspberry Pi 3B / 3B + / 4B, we made images for RaspberryPi.

ALT is an independently developed distribution popular in Russia which is available in Workstation and Education editions (https://en.altlinux.org/Downloads). There is also a special non-commercial version of the distribution called "Simply Linux" https://en.altlinux.org/Simply_Linux_9

https://getalt.org/en/alt-workstation/ ALT Workstation 9.1 (Raspberry Pi 4/3)
https://getalt.org/en/alt-education/ ALT Education 9.1 (Raspberry Pi 4/3)
https://getalt.org/en/simply/ Simply Linux 9.0 (Raspberry Pi 4)

Could you add images of ALT distributions to the Raspberry PI image download page (ie https://www.raspberrypi.org/downloads/), as well as in the "Raspberry Pi Imager"?
We offer this new section (for Raspberry Pi Imager):
```
       {
            "name": "ALT",
            "description": "Choose from ALT images",
            "icon": "https://getalt.org/raspi/logo-alt-ext.png",
            "subitems_url": "https://getalt.org/raspi/os_list_imagingutility_alt.json"
        },
```
Link to os_list: https://getalt.org/raspi/os_list_imagingutility_alt.json
The "rpi-imager" package is compiled into our package repositories (https://packages.altlinux.org/en/sisyphus/srpms/rpi-imager). Thank.

gochankot, 2020-08-05:
ping @maxnet @lurch

lurch, 2020-08-05:
I think this might be a @ghollingworth decision, and he's currently on holiday...

jqt4, 2021-03-10:
I would like to remind you about adding the ALT distribution images to the "Raspberry Pi Imager".
ALT is a distribution with complete and well-executed Russian localization, which will facilitate the use of RPi by Russian users.

jqt4, 2021-03-15:
@ghollingworth

Could you help me.
Please see my comment above and add an Alt section to the "Other general purpose OS" section.

Thanks.

ghollingworth, 2021-03-17:
The top-level description needs to be more descriptive, such as "ALT Linux: Russian language-specific operating system"

Also, it seems your JSON file contains three operating systems one of which is not ALT Linux, you are not allowed to add random extra operating systems to your JSON file.

jqt4, 2021-03-19:
Thanks a lot for your reply.
> The top-level description needs to be more descriptive, such as "ALT Linux: Russian language-specific operating system"

Yes, that's better.
Please add the following new subsection (in "Other general purpose OS" section):

       {
            "name": "ALT",
            "description": "Russian language-specific operating system",
            "icon": "https://getalt.org/raspi/logo-alt-ext.png",
            "subitems_url": "https://getalt.org/raspi/os_list_imagingutility_alt.json"
        },
        

> 
> Also, it seems your JSON file contains three operating systems one of which is not ALT Linux, you are not allowed to add random extra operating systems to your JSON file.

The name "Simply Linux" does not contain the word "ALT", however this distribution is released by the same organization BaseALT Ltd. as the rest.
Please see these pages:
https://en.altlinux.org/Simply_Linux_9
https://getalt.org/en/simply

Thanks.

XECDesign, 2021-03-24:
Added

closed 2021-03-24 by ghollingworth