raspberrypi/rpi-imager #99: Error running authopen to gain access to disk device 'dev/rdisk2'

wsm1992, 2020-08-04:
I want to write os to my sd card. After I click WRITE button and then popup an error "    Error running authopen to gain access to disk device 'dev/rdisk2'   ",
is it my setting problem or others?

maxnet, 2020-08-04:
You do get a prompt for your password, but after you entered it, you get this message?

The first time Imager tries to access storage Mac OS X will -after entering your password- provide you with a second prompt asking if Imager can have access to removable storage.
If you accidentally entered "no" there, Mac OS X will NOT prompt again.
(There is also another corner case in which Mac OS X may not prompt at all. But if you are a normal end-user, you should not experience that).
In that case you can change the settings under Preferences and settings -> privacy -> files and folders.

joelbrock, 2020-11-27:
same issue.

macOS V 10.15.5

I manually added Rpi Imager to System Prefs > Security & Privacy > Privacy > Full Disk Access but same error.

maxnet, 2020-11-27:
>I manually added Rpi Imager to System Prefs > Security & Privacy > Privacy > Full Disk 
>Access but same error.

Try giving access to "removable storage" under the tab "files & folders" instead.

joelbrock, 2020-11-28:
thanks for the reply!  that option isnt available in Cataline (10.15.5). I seem to only be able to add Applications to the full disk access panel.

On my laptop running Big Sur 11.0.1 i get the same error but i do see the privacy settings you mentioned.  They are already checked.

maxnet, 2020-11-28:
>thanks for the reply! that option isnt available in Cataline (10.15.5).

With "not available" you mean the entire "files and folders" tab is missing for you?

- It certainly do is supposed to exist in Catalina as well.
Here is a screenshot I took of it earlier this year (before I upgraded):

[image]

- You do are certain you do have Catalina and not something older?

- You did not get any prompt to allow access to "removable storage" when you first used Imager either?
Looks like this:

[image]

- The computer does is your private property? As opposed to belonging to an employer that could theoretically have restricted your access to certain settings?

- You may also want to try if you can get "Files & Folders" to open programatically by going to "Finder" -> "Applications" -> "Utilities" -> "Terminal" and typing: 
```
open "x-apple.systempreferences:com.apple.preference.security?Privacy_RemovableVolume"
```

maxnet, 2021-01-17:
"Error running authopen" was changed into a more user-friendly error message pointing to the privacy & security settings in 1.5, so closing this one.
If anybody still has issues with this, open a new issue.

closed 2021-01-17 by maxnet