raspberrypi/rpi-imager #102: [Feature Request] Add tick box to enable SSH

Jake-Barwell, 2020-08-13:
With the ability for Raspbian to detect a file in the boot sector and enable ssh, I believe it would be useful for beginners to be able to check a box to have this file automatically included. 

This would simplify the process, mainly on windows where the ability to change a file extension requires changing settings in file explorer. Most new users would most likely not know about this feature anyway, so this checkbox would likely be the first time they encounter SSH on the PI.

Jake-Barwell, 2020-08-13:
another potential use case for this type of customization would be the ability to enter WiFi credentials and have the program compile a WPA supplicant file.

ghollingworth, 2020-08-13:
Duplicate of many others

closed 2020-08-13 by ghollingworth