raspberrypi/rpi-imager #103: [OS Request] TwisterOS

mobilegmYT, 2020-08-26:
Hello!
I'd like to ask if TwisterOS could be added to the list of OS's available for the imager. It is based on rpiOS and is very stable.  It is also a little more friendly for new pi users because of its familiar GUI. This is the website link: [https://twisteros.com/index.html](url)

sysdl132, 2020-09-26:
Hey @mobilegmYT !
You can look at https://downloads.raspberrypi.org/os_list_imagingutility.json , make your own and upload to your own file server.Then edit config.h(what to edit: look at comments) and build the imager.

lurch, 2020-09-26:
No need to edit config.h and rebuild, just use the `--config` command-line flag, see https://github.com/raspberrypi/rpi-imager#custom-repository  :slightly_smiling_face:

mobilegmYT, 2020-09-26:
Sorry if i sound a little noob, but I launched it with `--repo https://archive.org/download/twister-osv-1-8-2.img/TwisterOSv1-8-2.img.xz
` and nothing showed up on the choose os menu. I'm pretty sure I did something wrong.

lurch, 2020-09-27:
You need to point it at the URL of a JSON file (see https://downloads.raspberrypi.org/os_list_imagingutility.json for an example), not the URL of your image file.

mobilegmYT, 2020-09-27:
Ok. So I got that part working, the last problem I have is I don't really understand the format of image_download_size and extract_size. It's certainly not in bits, but not kilobits.

mobilegmYT, 2020-09-28:
Last question. Would it matter if I removed things like the checksum and file size of my OS from the json file?

lurch, 2020-09-29:
That's a question for @maxnet , but I _suspect_ that RPi Imager probably uses those to verify that the download was successful and didn't get corrupted? Why would you not want to add them?

mobilegmYT, 2020-09-29:
Cuz I couldn't figure out what they did and I asked here, but no one replied so 🤷‍♂️. I had to finish the thing on a deadline lol.

mobilegmYT, 2020-12-08:
> No need to edit config.h and rebuild, just use the `--config` command-line flag, see https://github.com/raspberrypi/rpi-imager#custom-repository 🙂

Well... I tried doing that on my RPI by compiling with `debuild -uc -us --config https://raw.githubusercontent.com/mobilegmYT/TwisterOS-imager/master/os_list_imagingutility.json`, but it told me that --config wasn't a valid flag.

maxnet, 2020-12-08:
`--repo` not config.

mobilegmYT, 2020-12-08:
Ah ok. I was wanting to compile a version instead of launching with the repo command because on platforms like windows, switching the desktop shortcuts to add the repo flag is pretty hard. I think editing config.h and recompiling should make it automatically use my preferred json file without adding the repo flag right?

maxnet, 2020-12-08:
Yes, can also change the repo in `config.h`
But then you probably want more changes as well, to change the name of the program, have it installed in different location, etc.

mobilegmYT, 2020-12-08:
Yes

lurch, 2020-12-09:
> switching the desktop shortcuts to add the repo flag is pretty hard

It's not that hard, @procount shows an example of doing that here: https://github.com/procount/pinn/issues/442#issuecomment-737782216

Alternatively, I guess you could just create a `flash_twisteros.bat` file for people to download? :wink:

mobilegmYT, 2020-12-09:
Well nobody wants to edit files to flash an OS right? I'm trying to compile the installer with the readme steps but the Qt online installer says it "Can't connect to the server". Maybe I'll try the offline installer and see how that goes.

lurch, 2020-12-09:
Just to play devil's advocate: nobody wants to install a custom version of RPi Imager just to install TwisterOS, when they already have a perfectly functional version of RPi Imager already installed, right? :wink: :rofl: 

(I'm just kidding, you're of course welcome to create your own custom version if you really want to)

mobilegmYT, 2020-12-09:
Most people who are installing TwisterOS are noobs to RPI, and it would be much easier for them to download the program and click a few buttons then to have to download the image do and a bunch of stuff with etcher 😉. A lot of people also download the patches instead of the img file for some reason, so it would also clear up some confusion.

ghollingworth, 2020-12-10:
If you wait for a couple of weeks we should be able to add TwisterOS to the rpi-imager standard set (we're currently working on implementing sub-menus), which should mean you can then point people at the standard Raspberry Pi imager ..

It's just taking longer than expected to get here, but we've at least got a release ready for testing now...

Gordon

procount, 2020-12-10:
Cool! 😎 
Does that mean I'll be able to have a sub-menu for PINN?

ghollingworth, 2020-12-10:
Possibly, although I'm not sure how it'll fit into the menu structure, or how the description will communicate what PINN is.

mobilegmYT, 2020-12-10:
Well at least I don't have to compile my own modified version of the imager now. Was banging my head on openssl and cmake the other day.

mobilegmYT, 2020-12-30:
> If you wait for a couple of weeks we should be able to add TwisterOS to the rpi-imager standard set (we're currently working on implementing sub-menus), which should mean you can then point people [...]

I might be wrong, but aren't sub-menus already in it?

Itai-Nelken, 2020-12-31:
yes they already are (as of v1.5 I think).
any ETA to when TwisterOS will be added?

mobilegmYT, 2021-01-19:
This issue is inactive, so I'll close it and try to make a PR with a TwisterOS submenu.

Itai-Nelken, 2021-01-27:
ETA to when it will be added?

if I understood correctly, here: https://github.com/raspberrypi/rpi-imager/issues/103#issuecomment-742338171 was said that it will be added after submenus where added, and they were added in version 1.5.

mobilegmYT, 2021-01-27:
In a few hours, the Twister OS submenu should be online at https://twisteros.com/os_list_imagingutility_twisteros.json. I'll notifiy you when it's up so it can be added to the main OS list.

mobilegmYT, 2021-01-27:
@ghollingworth Hello! Here is a finished submenu. Can you see about getting this into the master list rpi-imager uses in the other desktop OSes section? https://raw.githubusercontent.com/mobilegmYT/TwisterOS-imager/master/os_list_imagingutility_v2.json

ghollingworth, 2021-01-28:
Where's you license statement?  Where are the sources, it says it is open source but I cannot find the sources for the patches.  Also it seems it is using images from Microsoft and Apple, do you have licenses for those?

Itai-Nelken, 2021-01-28:
> Where's you license statement? Where are the sources, it says it is open source but I cannot find the sources for the patches. Also it seems it is using images from Microsoft and Apple, do you have licenses for those?

@ghollingworth I think all Apple and Microsoft images were removed.

closed 2021-12-09 by ghollingworth