raspberrypi/rpi-imager #104: Release date attribute missing in metainfo file

ghisvail, 2020-08-30:
The metainfo file is no longer valid. The recent [change](https://github.com/raspberrypi/rpi-imager/blob/c08887f788bf1e67d246403d61dd5025e2b4765d/linux/rpi-imager.metainfo.xml.in#L58) to template it removed the mandatory release date attribute.

maxnet, 2020-08-30:
Is the requirement documented somewhere?
As other projects like VLC leave it out as well...

maxnet, 2020-08-30:
https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html#qsr-app-contents-suggestions

==
4.1.4 Suggested metadata file contents #
It is useful to add these tags as well if they make sense for the described application. **They are not strictly required to be present though.**

releases
==

https://www.freedesktop.org/software/appstream/docs/chap-Metadata.html#tag-releases

==
A release tag **can** have the properties version, date and timestamp
==

ghisvail, 2020-08-30:
> A release tag **can** have the properties version, date and timestamp

And a couple of sentences after: "The **date property** can have any time in ISO 8601 format as its value and **should be present for every release**"

maxnet, 2020-08-30:
Would argue that in standards "should" means a recommendation, rather than a requirement.
Like is standardized in RFCs: https://www.ietf.org/rfc/rfc2119.txt

Release date is a bit a pain, as the application is compiled before it is decided it is good enough for release, and release date is known.

ghisvail, 2020-08-31:
> Would argue that in standards "should" means a recommendation, rather than a requirement.

Maybe.

> Release date is a bit a pain, as the application is compiled before it is decided it is good enough for release, and release date is known.

I understand your pain, though I don't think the metainfo file was meant to be templated originally. It's supposed to list the history of releases (with respective changelog for each entry ideally) instead of overwriting a single entry each time.

maxnet, 2020-08-31:
>Maybe.

Well, the only thing the validation tool that comes with Ubuntu complains about is about the hypen in rpi-imager, and that we have an empty content-rating listing:

```
$ appstreamcli validate rpi-imager.metainfo.xml 
I - rpi-imager.metainfo.xml:org.raspberrypi.rpi-imager:3
    The component ID [org.raspberrypi.rpi-imager] contains a hyphen/minus. Using a 
    hyphen is strongly discouraged, to keep interoperability with other tools such as 
    D-Bus. Ideally, replace any hyphens with an underscore ('_')

W - rpi-imager.metainfo.xml:org.raspberrypi.rpi-imager:60
    Found empty 'content_rating' tag.

Validation failed: warnings: 1, infos: 1
```

The empty content-rating sounds correct to me.
Guess applications that do not contain anything offensive, and do not sell your data are so rare nowadays that it thinks it may be a mistake :-)

What application is unhappy about our lack of release date?

ghisvail, 2020-08-31:
Flatpak uses a stricter validation tool, see this [build log](https://flathub.org/builds/#/builders/39/builds/3118). The same tool does not complain about the empty content rating tag (and it is correct, indeed).

tdewey-rpi, 2023-11-17:
Closing as stale. Please raise a new issue if you believe it still applies.

closed 2023-11-17 by tdewey-rpi